TheKikGen/MPC-LiveXplore #5: MPC Live - update image  analysis

mamatt, 2019-12-28:


mamatt, 2019-12-28:
Found  0x3C-0x4F : 

this is an SHA1 hash of the ext4.xz  file !

mamatt, 2019-12-28:
0x04 => filesize
0x0C => not sur but seems to be a string table  
0x24 => this is strange but is seems to be the adresse of the second byte of sha1 hash !

mamatt, 2019-12-28:
I can confirm this is good for 2.2 too ! even if ext4 partition is not compressed !

To summarize :
 - HEADER 
    - START : 0x0  
    - END   : 0x103
    - LENGTH : 0x104
 - FOOTER : 
     - START : 0x103 + IMG Size
     - LENGTH :  0xE0

Headers and footers are quite the same between different image except for those bits :
<pre>
HEADER
0x04 => int32 MPC-Update filesize 
0x0C => 4 bytes Address of what seems to be a strings table
0x24 => 4 bytes this is strange but is seems to be the address of the second byte of sha1 hash !
0x4C => date image (hour_t) 32bit
0xB8 => 8 bytes string image version
0xFC => Int32 image(payload)  filesize

FOOTER (Addr relative to Footer address)
0x3C-0x4F => SHA1 hash of the payload (ext4 or compressed ext4)
</pre>

With those infos we should be able to construct a valid MPX-2.X-update.img file and use if with InMusic/AkaiPro stuff to push modified image ;)

mamatt, 2020-01-22:
In fact the whole image is a dtb !

you can decompress it with : dtc -I dtb -O dts -o MPX-2.X-update.dts MPX-2.X-update.img

TheKikGen, 2020-01-23:
Nice ! I used an hex editor to save the compressed ext4 partiton.  It is xy compressed since 2.6.

TheKikGen, 2020-01-23:
For previous versions of MPC Live/x , it's here : https://www.akaipro.com/amfile/file/download/file/956/product/6/

TheKikGen, 2020-01-27:
```
/dts-v1/;
/ 
{
	timestamp = <0x5de0e899>;
	description = "MPC upgrade image";
	compatible = "inmusic,acv5", "inmusic,acv8";
	inmusic,devices = <0x9e8403a 0x9e8403b>;
	inmusic,version = "2.7.0.43";
	images {
		rootfs {
			description = "Root filesystem";
			data = <  (bynary data) >;
			partition = "rootfs";
			compression = "xz";
			hash {
				value = <0xba31c00 0x56e3a8f0 0x9a55019c 0x1e8f64c2 0xb173019b>;
				algo = "sha1";
			};
		};
	};
};.

```

plule, 2020-05-25:
That’s nice, using those tools I could create an usb-installable image with ssh enabled :)

I’ve made that inefficient ugly script to ease extraction of the partition and recreation of the image: https://gist.github.com/plule/67e1c542e2f16d92ddda0ac479010c39

TheKikGen, 2020-05-27:
Sure you can !  I will have a look.  Does your script generate also all the sha1 hash code ?  This could allow us to use the official Akai updater.

plule, 2020-05-27:
Not sure about the computer installer, but the MPC itself accepts it without complaining (not using fastboot or the internal debug port)

TheKikGen, 2020-05-28:
OK ! So it should work with the win/mac client tool. I will test later.
I added you script to the repo with your credit and original url.
"Pierre" , are you french ? (I'am)

plule, 2020-05-30:
Oui !

TheKikGen, 2020-05-31:
Ok ! J'ai amendé le script pour générer des images Force.
J'ai testé avec la procédure de mise à jour depuis l'USB et cela fonctionne
très bien.

mamatt, 2020-05-31:
Hello,

pourrais-tu utiliser mon pseudo à la place de mon nom complet dans le script stp ;)

BTW, je ne suis pas sur que diffuser les firmware live/force patchées soit bien vu par Akai  :S

Merci

TheKikGen, 2020-06-02:
OK. pas de souci pour ton pseudo.
Pour les images, on verra si Akai réagit.  Si c'est le cas, on arrêtera la
diffusion.
Linux est open source.

closed 2020-11-27 by TheKikGen