TheKikGen/MPC-LiveXplore #7: TKGL_ANYCTRL - Adding more Qlinks, like the MPC X

mungewell, 2021-01-07:
Nice work with the 'TKGL_ANYCTRL' hack.

It would be very interesting to see if the Live/One could accept midi from a 'dial' controller to let the user control all 16 qlinks (like MPC X) without the need to page through 4 at a time. This would/could make the onboard plugins a bit more playable in a live sense.

Would this perhaps need a different SysEx identification to turn off the paging behavior?

TheKikGen, 2021-02-04:
Very concrete use case ! 
I own both a Live and a X, so I'm a bit  frustrated sometimes when playing on my LIve....

When the standalone MPC application starts, it sends a universal sysex 'identity request', and the controller will answer with an hardware id, and your serial number.  If you have a MPC Live, this will switch the software in a "Qlink bank select" mode.  I already tried to fake the apps by replacing the mpc live sysex header by the x one : it started but the controller was stuck at a point.  This implies more investigations.

The second way is to manage the "qlink button" on the Live to set it to the right bank before sending the qlink value.  The "led on" 1-4  note on message (the way the MPC controller manages leds) helps us to know the current state.  I implemented that in one of the modules  in the Kikpad project :  [here](https://github.com/TheKikGen/kikpad/blob/a4be6090484048b5abbe8e181c7cca164cab1bd5/mod_kikpad_MPC.h#L56)   

````````
///////////////////////////////////////////////////////////////////////////////
// Set Qlink group by simulating Qlink select button press
///////////////////////////////////////////////////////////////////////////////
static void SetQLinkCurrentGroup(uint8_t group) {

  midiPacket_t pk;

  if (group == QLinkCurrentGroup || group > 3  ) return;

  if ( group > QLinkCurrentGroup )
    group -=QLinkCurrentGroup;
  else
    group = 4 - QLinkCurrentGroup + group;

  for ( uint8_t i = 0 ; i < group ; i++  ) {
    // Simulate QLink select button
    pk.packet[0] = 0x09 ;
    pk.packet[1] = 0x90 ;
    pk.packet[2] = 0X00 ;
    pk.packet[3] = 0x7F ;
    MidiUSB.writePacket(&pk.i);
    // Released
    pk.packet[3] = 0x00 ;
    MidiUSB.writePacket(&pk.i);
  }
}
````````
The last way but not the least, it to use a global midi mapping you define in the MPC software on your computer.  This is totally undocumented for the standalone mode.   I discovered while i was playing around controller midi traffic.  
Global midi mapping allows the mapping of all buttons, pads  and Qlinks.  So, you set a mapping file with the MPC PC software (ctrl + shift + L for learning mode), export the xmm mapping file and transfer it on the internal sdcard in a specific folder.  You need also to update the MPC.settings file.
[Check that wiki note ](https://github.com/TheKikGen/MPC-LiveXplore/wiki/MPC-global-midi-mapping-in-standalone-mode-how-to)

Potreba, 2022-10-13:
Hey @TheKikGen is there a way to address internal midi automation bus? I've added 'anyctrl' to DOER_LIST , but haven't been able to unlock this hidden midi bus. I'm looking to launch scene 1 + move arrow down on MPC live with footswitch.

closed 2023-03-31 by TheKikGen