TheKikGen/MPC-LiveXplore #8: Sharing .xmm template

mamatt, 2021-01-21:
I've read your wiki about global mapping and it's seems very interesting !

if  I understand well, we don't need to overwrite the loader, and we can use any controler .

I'd likte to try  with a controller of my own, but  my problem is that I don't own a windows machine with mpc software installed.
Can you please share a .xmm file so I can use it as a template ?

TheKikGen, 2021-01-28:
This is my own mapping file for the Kikpad project.
[KIKPAD.zip](https://github.com/TheKikGen/MPC-LiveXplore/files/5889856/KIKPAD.zip)

For the corresponding mapping buttons, check directly the source here : https://github.com/TheKikGen/kikpad/blob/master/mod_kikpad_MPCController.h

TheKikGen, 2021-02-04:
Wine ?

TheKikGen, 2021-02-16:
I have posted more details about the xmm format https://github.com/TheKikGen/MPC-LiveXplore/wiki/MPC-global-midi-mapping-in-standalone-mode-how-to