TheKikGen/MPC-LiveXplore #9: Problems with "Change Sequence" and "Record" on MPC ONE

LoopinLab, 2021-03-15:
Hi,

thanks a lot for this awesome contribution. Play, Stop, Track Mute and even the Looper commands work like a charm on my MPC ONE!
I only have problems with Change Sequence and Record. Both of them work in the MPC2 Software, but not on my MPC ONE. There's just no reaction if I send the MIDI Control Change.
Could you check out if you experience the same issues? Probably the problem is on my side, but the rest is working great, and I don't see  I'm doing anything different with those two MIDI mappings.

Thanks a lot and best regards, Michael (LoopinLab)

TheKikGen, 2021-03-19:
Can you post your xmm file ?

LoopinLab, 2021-03-19:
Sure, I'll be glad to. Here you go:
[oo_210319_mcMidiMapping.zip](https://github.com/TheKikGen/MPC-LiveXplore/files/6172745/oo_210319_mcMidiMapping.zip)

LoopinLab, 2021-03-24:
@TheKikGen Does it work like this, or should I paste it in clear text in this issue?

LoopinLab, 2021-04-13:
Hi - is this board still active?

LoopinLab, 2021-04-14:
I guess the relevant part is this:

  <pairing>
    <Target_ Target_control="61"/>
    <Mapping_ Mapping_type="2" Mapping_channel="7" Mapping_data1="42" Mapping_control="5"
              Mapping_reverse="0"/>
  </pairing>

and it works great on the MPC Software, but no reaction from the MPC ONE Device.

TheKikGen, 2021-04-20:
Yes for sure it is active, but I answer on my free time...check that note and try to map to another key....
https://github.com/TheKikGen/MPC-LiveXplore/wiki/MPC-global-midi-mapping-in-standalone-mode-how-to

TheKikGen, 2021-10-30:
Any success with change sequence ?

About the mapping it self : 

- Mapping_ Mapping_type="2" : means that you are using a CC command
- Mapping_channel="7" : on the midi channel 7
- Mapping_data1="42" : the CC is 42
- Mapping_control="5" : the midi value is absolute 

Everything seems fine.
Are you sure to send the CC 42 correctly from your midi device ?
Do you have any other mapping using CC 42 ?
Maybe the "Change sequence" must be used in context on which that function is enabled, e.g. in song mode ?

closed 2023-03-31 by TheKikGen