TheKikGen/MPC-LiveXplore #14: Question - Expanding Audio Driver support

mungewell, 2021-08-30:
Hi,
With MPC v2.10 class compliant audio interfaces are supported, I assume (have MPC Touch myself) that the hardware actually relies on the Linux driver system (ALSA??) to implement this.

There are a few project which go beyond the hardware supported by the kernel. How realistic is the dream that these could be built for the MPC line?

If you wanted a request/suggestion/target, how about this one?? Self contained, but limited driver:
https://github.com/Drumfix/motu-avb-usb

Cheers,
Simon.

TheKikGen, 2021-09-26:
This implies to Install the kernel source of your running kernel...but we don't have it....

mungewell, 2021-09-28:
Does the inbuilt kernel support loading modules (insmod, lsmod, modinfo commands)?

If so it may be possible to  build a new module based on the kernel's setup and then load that into it - ie without needing to rebuild the kernel itself.

TheKikGen, 2021-10-30:
Some modules can't be loaded as Akai customized the kernel for the MPC, and they are linked statically...
I tried 2 years ago to build a Kernel from scratch, but I missed a lot of specific drivers (like touch screen for example, or SoC board..).  I was successfull in launching a kernel at boot time, but only in command line mode !

Concerning Alsa, it is theoritecally possible to make a config file for the MOTU AVB, as the new 2.10 version support Alsa audio compliant sound cards. Did you tried to plug the board in your MPC ?

A fast track is probably to make it works on a Raspberry Pi, and then to make it works on the MPC.

henning, 2022-02-09:
This discussion goes down to something I already wondered recently since becoming aware that MPC is Linux based: 

Where are the sources that akai/inmusic is obliged to publish as result of using GPL code? 
Did anyone every ask them to provide it?
I found some GPL Notes on other inmusic Brand products (I think denonDJ if i remember correctly) bnut no info at all from akai and about MPC's.

Not sure if the GPL can enforce getting the kernel config used, but it might...

nicola-lunghi, 2024-06-28:
THis is an old thread but 
there's any possibility to change the sample rate of the usb interface to 48khz?
anyone has investigated how the usb sound card support works?

nicola-lunghi, 2024-06-29:
I don't think the issue is the kernel there's a dbus service 
az01-unsupp-dev
that is handling this I think.

here's the log from my allen heath cq18 (that is class compliant but works at 48khz) with 32ch in disabled

```
Jun 29 22:17:13 mpc-one-mk2 kernel: usb 2-1: new high-speed USB device number 6 using ehci-platform
Jun 29 22:17:14 mpc-one-mk2 kernel: usb 2-1: New USB device found, idVendor=22f0, idProduct=0023, bcdDevice= 0.01
Jun 29 22:17:14 mpc-one-mk2 kernel: usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Jun 29 22:17:14 mpc-one-mk2 kernel: usb 2-1: Product: CQ18T
Jun 29 22:17:14 mpc-one-mk2 kernel: usb 2-1: Manufacturer: Allen&Heath Ltd
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device add: 2-1 (driver 'usb'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device add: 2-1:1.0 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.0')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device bind: 2-1:1.0 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.0')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device add: 2-1:1.3 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.3')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device add: 2-1:1.4 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.4')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device bind: 2-1:1.3 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.3')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device add: 2-1:1.2 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.2')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device bind: 2-1:1.4 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.4')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device bind: 2-1:1.2 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.2')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device add: 2-1:1.1 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.1')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device bind: 2-1:1.1 (driver 'snd-usb-audio'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1/2-1:1.1')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: device bind: 2-1 (driver 'usb'; syspath '/sys/devices/platform/ff500000.usb/usb2/2-1')
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: waiting for queue to empty before signalling (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: signalling unsupported devices (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: delaying 190ms before signalling unsupported
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: signalling unsupported devices (1 pending)
Jun 29 22:17:14 mpc-one-mk2 az01-unsupp-dev[288]: skipping supported device 2-1:1.1 (driver snd-usb-audio)
Jun 29 22:17:14 mpc-one-mk2 az01-launch-MPC[323]: Reducing RLIMIT_DATA from 1913356288 -> 1913331712
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]: Error: 44100 may not be a supported sample rate on the external device.
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]:  Code: -22
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]: Error: 44100 may not be a supported sample rate on the external device.
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]:  Code: -22
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]: **** Failed to open ALSA device for capture hw:2
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]: **** Audio 44100Hz; 2-in; 2-out; 192sample buffer
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]: **** Warning: inefficient input path: hardware=2 filter=32
Jun 29 22:17:29 mpc-one-mk2 az01-launch-MPC[323]: **** Warning: inefficient output path: hardware=2 filter=32
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]: Error: 44100 may not be a supported sample rate on the external device.
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]:  Code: -22
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]: Error: 44100 may not be a supported sample rate on the external device.
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]:  Code: -22
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]: **** Failed to open ALSA device for capture hw:2
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]: **** Audio 44100Hz; 2-in; 2-out; 192sample buffer
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]: **** Warning: inefficient input path: hardware=2 filter=32
Jun 29 22:17:45 mpc-one-mk2 az01-launch-MPC[323]: **** Warning: inefficient output path: hardware=2 filter=32
```