TheKikGen/MPC-LiveXplore #15: MIDI map not preserved after restart

dmoracze, 2021-10-04:
Hello - I have an MPC One.

I have followed all of the instructions to get ssh set up, editing the `.xmm` file, and the settings file.
However, none of the mappings are working. I noticed that after I restarted the MPC and download the settings file, the custom path I entered is no longer there. It returns to:

```
  <VALUE name="MidiLearnEnabled" val="1"/>
  <VALUE name="MidiLearnPreviousMapping" val=""/>
```

Is there something else I need to do so that the MPC does not overwrite the settings file on start up?

Another, unrelated question. I'm a bit confused about how to implement the mapping. Take, for example:
```
<!--  Mode next sequence   -->
  <pairing>
    <Target_ Target_control="101"/>
    <Mapping_ Mapping_type="2" Mapping_channel="7" Mapping_data1="60" Mapping_control="5"
              Mapping_reverse="0"/>
  </pairing>
```
Does this mean I need to send a MIDI CC of 60 on channel 7 with a value of 5 to enter the "next sequence" mode without pushing the button on the MPC? How would I go about changing to the next sequence via MIDI. Say if sequence 1 is playing, I send a CC to change to sequence 2 after sequence 1 is done, then send the same CC to change to sequence 3 after 2, and so on....

Thanks for doing this work!

mungewell, 2021-10-05:
I believe that the `Mapping_control="5"` sets how the CC value will be interpreted.

If you set it to '2s comp' (`Mapping_control="7"`) you can have the setting increase by 1 each time the 'CC with value 1' is sent.

I was experimenting with changing the sequence with:
```
<pairing>
<Target_ Target_control="61"/>
<Mapping_ Mapping_type="2" Mapping_channel="1" Mapping_data1="5" Mapping_control="7" Mapping_reverse="0"/>
</pairing>
```

Unfortunately this immediate changes the sequence 'now', rather than at end of seq (Like 'shift-ctrl-alt-]' rather than 'shift-ctrl-]'). It also does NOT work whether 'next-seq' mode is selected/active.... the two operations seem to _fight_ each other.

TheKikGen, 2021-10-30:
Open an SSH session and stop the MPC app with "systemctl stop inmusic-mpc",
Modify your settings file, then reboot with a "reboot" command.  
The setting file is saved before the app is closed.

Potreba, 2022-05-03:
> Open an SSH session and stop the MPC app with "systemctl stop inmusic-mpc", Modify your settings file, then reboot with a "reboot" command. The setting file is saved before the app is closed.

Sorry, so is there a way to use app alongside hardware? I've managed to make cool mapping for mpk mini mk3, but it sometimes sets MPC.settings file back to default  <VALUE name="MidiLearnEnabled" val="1"/>
  <VALUE name="MidiLearnPreviousMapping" val=""/> 
  and I'm not sure why it edits MPC.settings back to <VALUE name="MidiLearnEnabled" val="0"/>

Potreba, 2022-05-03:
> Open an SSH session and stop the MPC app with "systemctl stop inmusic-mpc", Modify your settings file, then reboot with a "reboot" command. The setting file is saved before the app is closed.

@TheKikGen sorry to tag you and taking your time, but I think you're only person, who can clear things out.
've managed to make cool mapping for mpk mini mk3, but it sometimes sets MPC.settings file back to default

and I'm not sure why it edits MPC.settings   <VALUE name="MidiLearnEnabled" val="1"/> back to <VALUE name="MidiLearnEnabled" val="0"/>

And I didn't understand how app interacts with standalone in terms of MPC.settings file. Also I couldn't figure out what "Open an SSH session and stop the MPC app with "systemctl stop inmusic-mpc",
Modify your settings file, then reboot with a "reboot" command.
The setting file is saved before the app is closed." mean. Is it linux command? Is there some setting, to make MPC.settings permanent? Maybe lock the file to prevent overwriting?

TheKikGen, 2022-05-04:
I recommend to save your settings file once it's ok.
I don't know exactly when it could be resetted.  I suppose if you change to deeply your settins in the ad hoc page, that could affect your "global mapping" that is unsupported officially.
The settings file is saved when the MPC app is closed.
So to be sure that your mapping will be taken in account stop app before then edit the settings file
# systemctl stop inmusic-mpc
(edit the settings file)
# systemctl restart inmusic-mpc

EscapeTheBoom, 2022-08-30:
I just tried to update the MPC.settings after updating to 2.11.2.2 and it drove me crazy. After a reboot, MPC always changed the MidiLearnPreviousMapping back to some other .xmm in the /usr path (and always a different one).
Downgraded it back to 2.10.1.83 and it was possible again. Maybe that helps, @Potreba 
Seems like AKAI put in an additional check of the file? 
@TheKikGen Do you see any way to avoid this?

arcin, 2022-09-07:
Just wanted to add my solution to this issue on 2.11. I've been running the 2.11 build for **3 weeks** and haven't had any more issues.

I recently bought a Midi Fighter Twister specifically for use with my mpc live 2.
Quickly noticed that AKAI must have patched something because there was defensive behavior against
1. enabling midi learn in `MPC.settings`
2. using a custom mapping in `MPC.settings`.

After *every single reboot* the `MPC.settings` file would have midi learn **disabled**. Also, my mapping would be overwritten with a default akai mapping. So if I changed my settings file to contain this

```xml
  <VALUE name="MidiLearnEnabled" val="1"/>
  <VALUE name="MidiLearnPreviousMapping" val="/media/az01-internal-sd/MPC Documents/Midi Learn/MyCoolMapping.xmm"/>
```
Restarting the mpc or changing settings via the mpc's menu would change those lines in `MPC.settings` to
```xml
<VALUE name="MidiLearnEnabled" val="0"/>
<VALUE name="MidiLearnPreviousMapping" val="/media/az01-internal-sd/MPC Documents/Midi Learn/Novation Launchpad Mk2.xmm"/>
``` 

I managed to get midi learn to stay enabled by using the `reboot` option recommended by @TheKikGen. The surprising thing was that my mapping kept getting changed to
```xml
<VALUE name="MidiLearnPreviousMapping" val="/media/az01-internal-sd/MPC Documents/Midi Learn/Novation Launchpad Mk2.xmm"/>
```

So I thought *"no big deal, I'll overwrite it"*.
> Replace `mpclive` here with your mpc's ip address.
> Also replace `MyCoolMapping.xmm` with your mapping file.
```bash

# stop the mpc app and remount the drive as read/write
# So that you can overwrite the built in mappings
ssh root@mpclive systemctl stop inmusic-mpc
ssh root@mpclive mount remount -o rw,remount /

# Overwrite the built in Akai mapping.
scp ./MyCoolMapping.xmm \"root@mpclive:'/usr/share/Akai/SME0/Midi Learn/Novation Launchpad Mk2.xmm"
# remount the drive as read-only
ssh root@mpclive mount remount -o ro,remount /
# restart the mpc app
ssh mpclive systemctl restart inmusic-mpc
```

After booting up, the `MPC.settings` file changed the mapping line again. To
```xml
<VALUE name="MidiLearnPreviousMapping" val="/media/az01-internal-sd/MPC Documents/Midi Learn/Novation Impulse 61.xmm"/>
```
I overwrote that file and again, after a boot up the line was changed to another mapping. I didn't figure out the pattern exactly but **Akai definitely added something to cycle through mappings if the file is changed.**

Maybe they're checking dates or a checksum or extended attributes. I don't know, but at that point I was pretty frustrated and decided to just overwrite all of them because they're not useful in standalone anyway. Here's the code.

> Again, replace `mpclive` with your mpc's ip address.
```bash
# Copy all the AKAI mappings from your mpc to your computer.
mkdir Midi\ Learn
scp -r "root@mpclive:'/usr/share/Akai/SME0/Midi Learn'" ./

# stop the mpc app and remount the drive as read/write
# So that you can overwrite the built in mappings
ssh root@mpclive systemctl stop inmusic-mpc
ssh root@mpclive mount remount -o rw,remount /

# List all akai mappings, filter out any file that isn't a midi mapping, and replace every Akai mapping with the contents of yours.
ls Midi\ Learn | grep ".xmm" |  while read line; do scp "MyCoolMapping.xmm \"root@mpclive:'/usr/share/Akai/SME0/Midi Learn/$line'\""; done

# remount the drive as read-only
ssh root@mpclive mount remount -o ro,remount /
# restart the mpc app
ssh mpclive systemctl restart inmusic-mpc
```
After that, everything worked fine. I wrestled with this problem for a couple of hours before resorting to overwriting every built in mapping but it doesn't seem like Akai added any guards against this.  My midi fighter twister mapping has been working in standalone without issue. I can freely connect other controllers and do project specific mappings in standalone and have my global mapping work alongside that. 

Anyway, this post is way too long but I wanted to be specific incase it saves someone the trouble of figuring it out. :+1:

### NOTE
I did have to use `ssh root@mpclive reboot` instead of `ssh root@mpclive systemctl restart inmusic-mpc` when I first got midi learn to stay enabled. I know that doesn't make much sense and it's possible I'm misremembering, was too sleepy, or was just too in the weeds when debugging. But, if you haven't tried that give it a try.

```bash
ssh root@mpclive systemctl stop inmusic-mpc
# ... edit settings file on mpc or scp yours from your computer to the mpc.
ssh root@mpclive reboot
```

EscapeTheBoom, 2022-09-08:
Thank you very much, @arcin - this is really helpful and I am excited to give it a try soon. 
@TheKikGen - do you see any possibility to make this process easier again in a future version of your ssh-images? That would be awesome.

EscapeTheBoom, 2022-09-14:
Just wanted to let you know it worked perfectly on my MPC ONE as well. Thank you @arcin - your description was incredibly helpful. 👍

technocybernaut, 2023-02-24:
Hi, I hope someone here might have some insight on troubleshooting this further. First, I greatly appreciate all the work everyone has contributed to make this happen. Thank you!

So I have followed the original steps and the steps outlined by @arcin with using the reboot command instead but am still having trouble keeping the "MidiLearnEnabled" value to stay "1" and not revert back to "0". I have tried to be methodical and notated when the value reverts if that helps?

-Running 2.11.8 SSH firmware
-It really only stays preserved with the MPC set to open to a blank project on startup, shut down, and reboot. 
-Touching a knob on the SL makes it instantly revert to "0". Qlinks can be touched and it stays active.
-Opening another project makes it reverts. I even tried this with a project I specifically made and saved while the setting was active.
-Midi CCs from the SL are still picked up in Midi Programs.
- Using an old version of OSX and finder wasn't working. Using both Terminal and Termius to connect. 
- My .XMM file was created in the desktop software custom for an Novation SL mkIII 49 and exported.
- I added the .XMM file to the applicable folder and the pathway is saved in the "MidiLearnPreviousMapping" field, minus a few times while troubleshooting different "MPC.settings" files which were exported and edited in different ways. The MPC made a file called something like "My Midi Map.XMM" on it's own and switched the path to that. Not since.
-Side note, I am finally doing the hack because I only just found the fix for the 3+min boot and project load times issue. Thousands of duplicate files in the "Midi Learn" folder. I was on 2.11.6 at the time and deleted the entire folder as I had not yet tried SSH access. I had the issue through 2.10.

Does anyone have any ideas on what else I might try?

sean808080, 2024-01-13:
Hello all..what’s the status of this feature? Being able to map a midi controller once and have it persist across projects would be very useful indeed.  Has anyone got it working reliably ? If so, should this issue be closed?