TheKikGen/MPC-LiveXplore #19: How to add my custom systemd service?

davidmoreno, 2022-01-17:
Hi, 

I managed to get initial rtpmidi support into the MPC One (issue tracking the progress https://github.com/davidmoreno/rtpmidid/issues/81, [binary](https://github.com/davidmoreno/rtpmidid/files/7884696/rtpmidid.gz)). So far manually running a statically compiled rtpmidid library without mdns / avahi / zeroconf support (so only directo connection by ip). 

I was thinking that this is useful enough to add to my rooted MPC one.. but how can I install a systemd service? Do I need to crete a new image? or can I mount read write temporarily the filesystem? 

Just adding two files is enough (the binary in for example /usr/local/bin) and a .service file similar to this [this](https://github.com/davidmoreno/rtpmidid/blob/master/debian/rtpmidid.service) at `/etc/systemd/system/'.

BTW rtpmidi in MPC one is amazing! Super low latency and huge posiblities!

davidmoreno, 2022-01-17:
I answer myself:

At MPC: 
```sh
mount -o remount,rw /
mkdir /tmp/root
mount --bind / /tmp/root
```

At my computer. Prepare the rtpmidid file, and edit rtpmidid.service to do not use the user, nor working directory:

```sh
scp rtpmidid mpc-one.local:/usr/bin/
scp rtpmidid.service mpc-one.local:/tmp/root/etc/systemd/system/
```

Back at MPC:

```sh
ln -s /etc/systemd/system/rtpmidid.service /tmp/root/etc/systemd/system/multi-user.target.wants/
umount /tmp/root/
mount -o remount,ro /
```

Reboot, rtpmidi!

So.. solved. So I will change my question:

1. Where can I document this to make a pull request so next that tries knows how to do it?
2. Is it an option to have rtpmidid bundled with the images? If not tell me and I will prepare an installer.

henning, 2022-01-17:
@davidmoreno this sounds like very great work - thank you! I just found this project on my search for any information on the MPC's network midi protocol and its funny that a few days after I started watching it you come along with a solution that I'm interested in ;)

DatanoiseTV, 2022-01-18:
@davidmoreno It is possible to bundle it with a custom image, but I am not sure if it is a good idea. I would prefer the way of an installer script.

TheKikGen, 2022-01-20:
@davidmoreno super job !
The best place is in the WIKI.

davidmoreno, 2022-01-20:
Got it!

It uses PATH and LD_LIBRARY_PATH, creates an user, and copy some files to make avahi work, and then plain rtpmidid works.

To use it, uncompress the attached tar.gz file to just a tar file, copy the file to the MPC and:

```
tar fx mpc-one-rtpmidid.tar
cd mpc-one-rtpmidid
./test.sh
```

It will work until close terminal or reboot.

I had some problems with the installer, if I can advance on this will keep an installer too.

BTW, if someone knows somebody a AKAI, or somebody from AKAI reads this, please consider just using rtpmidid. Just 25 files and rtpmidi is available and just works! No even need to change anything at the UI.

[mpc-one-rtpmidid.tar.gz](https://github.com/TheKikGen/MPC-LiveXplore/files/7908856/mpc-one-rtpmidid.tar.gz)

TheKikGen, 2022-02-09:
I investigated a little more on the subject.

The dbus rules forbid the ownership of avahi service, and to avoid conflicts, it is necessary to stop the network midi services with  `systemctl stop az01-network-midi`

Then adding a local rules file at /media/az01-internal/system/etc/overlay/system-local.conf

```
<!DOCTYPE busconfig PUBLIC
"-//freedesktop//DTD D-Bus Bus Configuration 1.0//EN"
"http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>
    <policy user="root">
        <allow eavesdrop="true"/>
        <allow eavesdrop="true" send_destination="*"/>
    </policy>
  <policy context="default">
    <allow own="org.freedesktop.Avahi" />
  </policy>
</busconfig>
```

(the root user policy allow bus sniffing)

TheKikGen, 2022-02-09:
The avahi customized by InMusic is /bin/DeviceControlServer.bin

TheKikGen, 2022-02-10:
Nb : at boot time, only the rootfs etc and its overlay at az01-internal are checked for config files.
Critical point : if the config file syntax is bad, the dbus will not start : that means a BOOT LOOP sequence...
It is why it is much better to copy dbus rules files on the image, because it can be resflashed easily with the update procedure.