TheKikGen/MPC-LiveXplore #21: small issues in .profile

egandro, 2022-02-07:
There are small issues in the .profile.

1) why do we have a $ befor the name?
2) hostname -i isn't a foolprove way to getting the IP

Try this :) 

```
ip a s|sed -ne '/127.0.0.1/!{s/^[ \t]*inet[ \t]*\([0-9.]\+\)\/.*$/\1/p}'
```

Credits for that: https://stackoverflow.com/questions/13322485/how-to-get-the-primary-ip-address-of-the-local-machine-on-linux-and-os-x

```
$root@mpc-one:~# hostname -i
hostname: mpc-one: Unknown host
```

TheKikGen, 2022-02-07:
Thanks to point that !

The first $ is a mistake.  
I'm aware about "hostname"  resolution issue, but usually, the host name is resolved at /etc/machine-info for our mpcs.
I will use the sed command line from now...more reliable. You're right.

PS1='\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
echo "Your IP address is $(ip a s|sed -ne '/127.0.0.1/!{s/^[ \t]*inet[ \t]*\([0-9.]\+\)\/.*$/\1/p}')"

TheKikGen, 2022-02-07:
You can rapidly change that in the root/.profile by remoouting the root file system read write :
ssh root@(your ip) "mount -o remount,rw"  /
(via ssh to avoid polluting history)

and edit the file via an ssh browser (Thunar or WinSCP, Finder).

then remount in read only, 
ssh root@(your ip) "mount -o remount,ro"  /

alias ll='ls -l'
alias la='ls -a'
export PS1='\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
cat /root/.tkgl_logo
echo
echo "Welcome to the MPC OS ssh mod !"
uname -a
echo "Your IP address is $(ip a s|sed -ne '/127.0.0.1/!{s/^[ \t]*inet[ \t]*\([0-9.]\+\)\/.*$/\1/p}')"
echo "WARNING : you have root permissions. You could harm your system."

egandro, 2022-02-09:
Thx for the update!

closed 2022-02-09 by egandro