TheKikGen/MPC-LiveXplore #22: Midi over Ethernet question

johnsq, 2022-02-07:
First, please let me applaud your work here.  If I had not discovered MPC-LiveXplore, I would have sent my MPC Live II back. I take offence of being locked out of Global Midi by the manufacturer I handed my cash too. Not cool AKAI :(.

Anyway, I was wondering if you knew of a config change that would enable my MPC Live II to work with standard RTMidi protocols. Currently, as I'm sure you are aware, network midi is locked down to their own 'Akai Professional Network Midi' application, which runs only on Mac and Windows. If it were possible to turn off whatever authentication they use, I could happily connect network midi to my iPad.

Thanks

TheKikGen, 2022-02-07:
It's on my list....check another post here , and a solution built by David Moreno.
https://github.com/TheKikGen/MPC-LiveXplore/issues/19

Il will include that in a bootstrap module....when I'll have time :-) !

You can try to do that your self.  David has posted a zip file containing his rtpmidi hack.

TheKikGen, 2022-02-07:
Yes. I'm currently uploading somes updates of firmwares images.  Small adjustment but long uploads...

johnsq, 2022-02-08:
Thanks for your help @TheKikGen :)

TheKikGen, 2022-02-10:
rtpmidi is currently running on my mpc live...

closed 2023-11-30 by TheKikGen