TheKikGen/MPC-LiveXplore #23: Create build script to make custom img from original img

egandro, 2022-02-08:
I have suggestion.

Can we add a shell script (for nerds!) to create the custom firmware from the original img?

- start with "MPC-2.10.1-Update.img"
- check if the file is valid with sha256
- extract the file
- mount root.img
- copy files
- package the new file

Do you have such a builder somewhere in your project?

TheKikGen, 2022-02-09:
I have this script on my Linux dev platform, not in hte github, and it was more made on a table corner than to produce modified images in an industrial way ! 

You have summarized the steps quite well:

- Rootfs extraction with mpcimg
- Backup copy of the original rootfs image
- Mount the image on a local mnt folder
- Copy the bootstrap to mnt/usr/bin (check the bootstrap project here : https://github.com/TheKikGen/MPC-LiveXplore-bootstrap)
- Empty the system passwords
- Activate ssh
- Make new update image with mpcimg

egandro, 2022-02-09:
That is cool!

egandro, 2022-02-09:
Btw. Yesterday  I found out how to run a arm32/aarm64 Dockerfile on x86_64.

You have to install qemu and do some simple stuff (2 lines of code).

We can make a cross compiler plattform here.

The steps you explain here with copy stuff can be done without mounting. 

It would be awesome to have a non binary distribution system (for nerds).

I am trying to hack something up, as soon as I can diff the 2.10.0 orig and your image :)

TheKikGen, 2022-02-09:
I will post the diff for you ...

egandro, 2022-02-09:
Here my diff so far. 

`find . -type f -print0 | sort -z | xargs -r0 sha256sum ` in vanilla + your stuff and a diff to that

[diffs.txt](https://github.com/TheKikGen/MPC-LiveXplore/files/8035855/diffs.txt)

Is it possible for you to create two zip files and put it here in the repo?

Arp Patterns
and 
FX Racks

(just keep the directory structure)

If this is not possible (because of copyright issues - please let me know and we will find a solution for that)

egandro, 2022-02-09:
Sorry - maybe the question was stupid :)

Are that the files from here? https://github.com/TheKikGen/MPC-LiveXplore/tree/master/assets

egandro, 2022-02-09:
We also need to kill / replace this file ... and/or a builder option to set your own ssh file or your own root password.

**/root/.ssh/authorized_keys**

egandro, 2022-02-11:
My pull request is ready for review: https://github.com/TheKikGen/MPC-LiveXplore/pull/29

egandro, 2022-02-11:
It needs more testing / flashing.