TheKikGen/MPC-LiveXplore #24: xMPC-Homebrew Channel

egandro, 2022-02-08:
I would suggest creating something like the Wii/NDS/Switch has: A Homebrew Channel.

A simple Website where you can download packages to your AKAI.

More and more people are starting to create tar.gz files and It would be awesome to have a kind of "app" system.

We can use a web based software (http://mpc-foo/...) or a /homebrew installer directory on the SD Card.

Maybe we can take some software from https://github.com/fail0verflow/hbc/

They server can be a json file on github and tgz files

Potreba, 2022-05-03:
Nice daydreaming, but honestly it takes complete determination from musician to even install linux, not to mention something advanced. 

Anyway, out of all new things, I found few things REALLY useful: 
1. Force on MPC. This is huge, as in this historical moment it allows me to sidestep mpc's RAM limit and prepare live set. I use MPC live as a writing tool. And its really great, especially paired with midi controller of some sort
2. Global midi learn in standalone. This is important to me as I dont have to use dreaded menu button anymore - I just map pads on mini midi controller to different modes. This is really cool as it speeds up writing/ sound designing workflow immensely. 
3. SSH access is cool, as it allows to reorganize internal memory, copy some favorite patches, FX presets etc.

On the subject of daydreaming:
1. If plugin protocol could be opened to public we would likely see an MPC plugin market blossom. 
2. Global midi learn utility. Now it's kinda broken for me, as it works one time, but after few restarts it might not work again (mpc.settings file gets edited)

air-v, 2022-05-03:
Hello Potreba,

_If plugin protocol could be opened to public we would likely see an MPC plugin market blossom._

I think the code to develop your own plugin is not open because MPCs are more dedicated to live, and there is nothing more harmful than a crash due to a badly developed plugin in the middle of a session. Already it's boring when it happens to you in the middle of composing in my studio.... A plugin must be thoroughly tested before going into live mode.... But I agree that if it was more open one could imagine that other serious plugin company would do.

Cheers

closed 2023-11-30 by TheKikGen