TheKikGen/MPC-LiveXplore #25: Issue with mpcimg

egandro, 2022-02-08:
I am noticing an issue with mpgimg with the recent **MPC-2.10.1.tkglarp-Update.img**

I am trying to extract the rootfs from the image you made.

```
$ sha256sum ./MPC-2.10.1.tkglarp-Update.img
e916dcb55af16a0de08877090edb8481fd2434c53e6d0fbe227faec035b178da  ./MPC-2.10.1.tkglarp-Update.img
```

Unfortunately mpcimg fails:

```
 ./mpcimg extract ,/MPC-LiveXplore/MPC-2.10.1.tkglarp-Update.img xxx
---------------------------------------------------------------------
AKAI MPC/FORCE IMAGE TOOL - V1.1
https://github.com/TheKikGen/MPC-LiveXplore
(c) The KikGen labs.

NB : the dtc utility must be acessible from the current path !
---------------------------------------------------------------------

Extracting rootfs_xxx from ./MPC-LiveXplore/MPC-2.10.1.tkglarp-Update.img:

Akai image [./MPC-LiveXplore/MPC-2.10.1.tkglarp-Update.img] details :

timestamp            : 5dee18e1
description          : MPC upgrade image
compatible           : inmusic,acv5 inmusic,acv8 inmusic,acva inmusic,acvb inmusic,acvm
inmusic,devices      : 9e8403a 9e8403b 9e84046 9e84047 9e8404b
inmusic,version      : 2.10.1.83
hash value           : 82b4bbe8 45d166f4 dee26e2f 1cf6b5e4 4a0ae8d9

Roofs image is compressed with xz.
1/3.Extracting hexadecimal text data...please wait...
2/3.Generating binary data...
Size : 411889582
Traceback (most recent call last):
  File "./mpcimg", line 262, in <module>
    extract(sys.argv[2], "rootfs_" + sys.argv[3])
  File "./mpcimg", line 113, in extract
    binfile.write(bytearray.fromhex(wordstr))
ValueError: non-hexadecimal number found in fromhex() arg at position 16408
```

TheKikGen, 2022-02-09:
It works perfectly on my linux system with dtc 1.4.5 and python 2.7.17
Try to download again the update image.

TheKikGen, 2022-02-09:
Tested gain within MinGW64 with the image currently online :
```
---------------------------------------------------------------------
AKAI MPC/FORCE IMAGE TOOL - V1.1
https://github.com/TheKikGen/MPC-LiveXplore
(c) The KikGen labs.

NB : the dtc utility must be acessible from the current path !
---------------------------------------------------------------------

Extracting rootfs_test from MPC-2.10.1.83.tkglarp-Update.img:

Akai image [MPC-2.10.1.83.tkglarp-Update.img] details :

timestamp            : 5dee18e1
description          : MPC upgrade image
compatible           : inmusic,acv5 inmusic,acv8 inmusic,acva inmusic,acvb inmusic,acvm
inmusic,devices      : 9e8403a 9e8403b 9e84046 9e84047 9e8404b
inmusic,version      : 2.10.1.83
hash value           : 99284e23 7677c613 89c6dc98 1bb3a992 18d112fc

Roofs image is compressed with xz.
1/3.Extracting hexadecimal text data...please wait...
2/3.Generating binary data...
Size : 411853544
Pos  : 411853544
3/3.Decompressing rootfs img...please wait...
File  rootfs_test  ready in the current directory.
411853544  bytes written

Done !
```
DTC 1.6.0   and python : 3.8.9

egandro, 2022-02-09:
I am now on **27f3f4d7807b274b3179484136995c9f48a82e05b39b8f99b852588f4429658e**.

It worked!

Suggestion: please also provide a checksum file with your images :)

closed 2022-02-09 by egandro