TheKikGen/MPC-LiveXplore #26: Can using the ssh images brick the MPC? If so, how to unbrick?

henning, 2022-02-09:
I'm very interested in trying the hacks described here, clearly with a first step being to install the ssh images as described in  https://github.com/TheKikGen/MPC-LiveXplore/wiki/Enabling-SSH-on-the-MPC-Live-X-one-Force

I'm just a bit hesitant because I dont know what can possibly go wrong and what the rescue procedure is.

If things go weird and my MPC One would behave strangely, would I just simply run the update process with the latest official image, or can I get into situations where even that wont be possible anymore?
(assuming I dont go havoc with the power that I have when being ssh'ed as root on the machine and do crazy/stupid stuff there)

henning, 2022-02-09:
Probably found the answer in https://github.com/TheKikGen/MPC-LiveXplore-bootstrap

" if a script exists on a usb key at boot time. That's all. I use it everyday on my MPC Live and my MPC X. Upgrading your MPC or Force with one of these images will not brick your hardware. The update procedure is strictly the same as the one you use for the official images but of course, I can't be held responsible for any problem. You can update again to come back to an official image at any time."

might be interesting to have that information in place where the ssh image install is explained...

TheKikGen, 2022-02-10:
The ssh modded images will activate remote access in a console mode (ssh), and will add a tons of new arp patterns (check at the bottom of the list) + the test at boot that a script exists or not.  

Copy the image on a usb key , at the root , and choose the "Usb Drive Update" in the seettings/info/update menu

egandro, 2022-02-10:
@henning I am currently writing an opensource version of the custom firmware builder.

henning, 2022-02-11:
@egandro yes, I am observing your discussion, have subscribed to the whole project... I am just hesitant to start hacking on my MPC as I also love it for music making and dont wanna brick it. Actually thinking about getting a second one for these games...

air-v, 2022-02-11:
Don't be worry about bricking MPC. I've bricked my MPC ONE 3xtimes with bad img. The MPC start normaly with MPC Logo but after few second the display go to black screen and no buttons.To solve brick pb, you need to press  Full Level+Browse+Bank C and Power off. The System will boot in update mode (MPC Logo with usb signt on bottom screen).Connect your MPC on your PC thru USB and launch the update Firmware from USB. Flash it and tada.... MPC will be back to Official Firware.Regards

Herve Laporte

TheKikGen, 2022-02-11:
It is why we must set a principle : to not modify too deeply original Akai images.
Currently the only modifications are : activate ssh, liberate some restrictions on Dbus, arp patterns, fx racks bonus.

TheKikGen, 2022-02-12:
And i will remove the 3000 and more arp patterns in next releases of images. They stay available as a module in the bootstrap.

air-v, 2022-02-12:
What it could be smart with arp files is sort those files by categories if possible or maybe by folder  ....but i'm not sure that 's 
possible.

TheKikGen, 2022-02-13:
Not possible.  The MPC doesn't take about dir/sub-dir

closed 2023-03-31 by TheKikGen