TheKikGen/MPC-LiveXplore #30: FORCE Mode Loge appear and back to normal mode.

air-v, 2022-02-11:
Hi all, i've already done the SSH image on my MPC One. SSH connection is good and all sound good to.

When i run MPC with Force IMG on USB KEy, it boot on USB and Blue FORCE logo appears....And.......finish back to MPC soft. 
No FORCE mode at all. Somebody know why it doesn't start in FORCE mode ? Do i miss something ?
Cheers

TheKikGen, 2022-02-11:
This bug is currently corrected.
Image was rotated with the convert utility.

air-v, 2022-02-11:
Hi @TheKikGen,
You mean the last img on Github is working ? Because i've test it this morning and still not working ...
Maybe i need to be more patient...
Thans and again .....GREAT JOB for all

TheKikGen, 2022-02-11:
The last version of IamForce is not yet published. It will be during the week end I suppose.

zotanmew, 2022-02-15:
I just tested the latest IamForce version (I saw that you updated the link a few times) and sadly the same issue (force logo but boots into MPC mode) arises :(

zotanmew, 2022-02-15:
If I understand the logs and the config file correctly, the mapping fails because the config doesn't contain the live2 button/led IDs, I'll try to figure those out now

zotanmew, 2022-02-16:
Alright, I got it to work! Fixed all the button mappings that i could find. PRs: #31 and https://github.com/TheKikGen/MPC-LiveXplore-bootstrap/pull/3

TheKikGen, 2022-02-16:
> TheKikGen/MPC-LiveXplore-bootstrap#3

ok ! You mean this is working on a LIVE 2 !  If yes, could you post a photo
on the FB group ?

TheKikGen, 2022-02-16:
Best point to start is to copy paste the MPC LIVE example
(Map_LIVE_FORCE_ButtonsLeds
section), and rename the section to Map_LIVE2_FORCE_ButtonsLeds Live and
Live2 are close enough.

zotanmew, 2022-02-16:
Yep, I got it working! I don't have FB, I'll upload a picture here in a moment. To get it working I had to modify the regex in the mpcmapper library, see TheKikGen/MPC-LiveXplore-bootstrap#3

zotanmew, 2022-02-16:
Yep, I got it working! I don't have FB, I'll upload a picture here in a moment. To get it working I had to modify the regex in the mpcmapper library, see #31

air-v, 2022-02-18:
Hello Zotanmew,
Very promising. I still have the same pb as beginning with MPC One, starting with LOGO screen and back to MPC soft. I certainly should setup the conf file myself. How did you get the your live2 button/led IDs list ? i'll do it to my MPC One.
Regards
Herve

TheKikGen, 2022-02-20:
Note : the 220217 version if the iamforce had a bug in the configuration file parser.
Download the last correction on the homepage of the site and test again.

air-v, 2022-02-20:
Hello,
Try this morning with last iamforce img (only) .... 
Boot on nice KIKGEN Labs video but back on MPC file Menu and MPC soft.
Do i need to upgrade firmware too ?
Thx

TheKikGen, 2022-02-20:
It is linked with the  MPC One driver name (exactly the same issue as
Laura).
I'm on the patch now  !

TheKikGen, 2022-02-20:
@zotanmew 
Hi Laura.  Could post here the result of the 
`amidi -l `
on your MPC Live2 ?

Thanks;

zotanmew, 2022-02-20:
@TheKikGen tested with and without the force sdcard, seems to be identical:

```
root@mpc-live-ii:~# amidi -l
Dir Device    Name
IO  hw:1,0,0  MPC Public
IO  hw:1,0,1  MPC Private
IO  hw:1,0,2  MPC MIDI Port A
IO  hw:1,0,3  MPC MIDI Port B
```

TheKikGen, 2022-02-20:
@air-v (MPC One)  @zotanmew (Live II) 

Same request please but this time with 
`aconnect -l`

Thanks.

zotanmew, 2022-02-20:
```
root@mpc-live-ii:~# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 20: 'MPC Live II' [type=kernel,card=1]
    0 'MPC Public      '
        Connected From: 136:0
    1 'MPC Private     '
        Connecting To: 134:0
        Connected From: 135:0
    2 'MPC MIDI Port A '
        Connecting To: 137:0
        Connected From: 137:3
    3 'MPC MIDI Port B '
        Connecting To: 137:1
        Connected From: 137:4
client 130: 'Virtual MIDI Output 1 Input' [type=user,pid=299]
    0 'in              '
        Connected From: 137:7
client 131: 'Virtual MIDI Output 2 Input' [type=user,pid=299]
    0 'in              '
        Connected From: 137:5
client 132: 'Virtual MIDI Input 1 Output' [type=user,pid=299]
    0 'out             '
        Connecting To: 137:6
client 133: 'Virtual MIDI Input 2 Output' [type=user,pid=299]
    0 'out             '
        Connecting To: 137:2
client 134: 'Client-134' [type=user,pid=357]
    0 'TKGL Virtual In Private'
        Connected From: 20:1
client 135: 'Client-135' [type=user,pid=357]
    0 'TKGL Virtual Out Private'
        Connecting To: 20:1
client 136: 'Client-136' [type=user,pid=357]
    0 'TKGL Virtual Out Public'
        Connecting To: 20:0
client 137: 'MPC' [type=user,pid=357]
    0 'MPC Live II MPC MIDI Port A'
        Connected From: 20:2
    1 'MPC Live II MPC MIDI Port B'
        Connected From: 20:3
    2 'Virtual MIDI Input 2 Output out'
        Connected From: 133:0
    3 'MPC Live II MPC MIDI Port A'
        Connecting To: 20:2
    4 'MPC Live II MPC MIDI Port B'
        Connecting To: 20:3
    5 'Virtual MIDI Output 2 Input in'
        Connecting To: 131:0
    6 'Virtual MIDI Input 1 Output out'
        Connected From: 132:0
    7 'Virtual MIDI Output 1 Input in'
        Connecting To: 130:0

```

zotanmew, 2022-02-20:
@TheKikGen there you go :)

air-v, 2022-02-21:
Hello,
Here's mine 
root@mpc-one:~# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 20: 'MPC One MIDI' [type=kernel,card=1]
    0 'MPC Studio Live Public'
    1 'MPC Studio Live Private'
    2 'MPC Studio Live MIDI Port'
	Connecting To: 134:0
	Connected From: 134:2
client 130: 'Virtual MIDI Output 1 Input' [type=user,pid=294]
    0 'in              '
	Connected From: 134:5
client 131: 'Virtual MIDI Output 2 Input' [type=user,pid=294]
    0 'in              '
	Connected From: 134:3
client 132: 'Virtual MIDI Input 1 Output' [type=user,pid=294]
    0 'out             '
	Connecting To: 134:4
client 133: 'Virtual MIDI Input 2 Output' [type=user,pid=294]
    0 'out             '
	Connecting To: 134:1
client 134: 'MPC' [type=user,pid=266]
    0 'MPC One MIDI MPC Studio Live MIDI Port'
	Connected From: 20:2
    1 'Virtual MIDI Input 2 Output out'
	Connected From: 133:0
    2 'MPC One MIDI MPC Studio Live MIDI Port'
	Connecting To: 20:2
    3 'Virtual MIDI Output 2 Input in'
	Connecting To: 131:0
    4 'Virtual MIDI Input 1 Output out'
	Connected From: 132:0
    5 'Virtual MIDI Output 1 Input in'
	Connecting To: 130:0
root@mpc-one:~#

air-v, 2022-02-21:
Hello,
here's mine 
root@mpc-one:~# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 20: 'MPC One MIDI' [type=kernel,card=1]
    0 'MPC Studio Live Public'
    1 'MPC Studio Live Private'
    2 'MPC Studio Live MIDI Port'
	Connecting To: 134:0
	Connected From: 134:2
client 130: 'Virtual MIDI Output 1 Input' [type=user,pid=294]
    0 'in              '
	Connected From: 134:5
client 131: 'Virtual MIDI Output 2 Input' [type=user,pid=294]
    0 'in              '
	Connected From: 134:3
client 132: 'Virtual MIDI Input 1 Output' [type=user,pid=294]
    0 'out             '
	Connecting To: 134:4
client 133: 'Virtual MIDI Input 2 Output' [type=user,pid=294]
    0 'out             '
	Connecting To: 134:1
client 134: 'MPC' [type=user,pid=266]
    0 'MPC One MIDI MPC Studio Live MIDI Port'
	Connected From: 20:2
    1 'Virtual MIDI Input 2 Output out'
	Connected From: 133:0
    2 'MPC One MIDI MPC Studio Live MIDI Port'
	Connecting To: 20:2
    3 'Virtual MIDI Output 2 Input in'
	Connecting To: 131:0
    4 'Virtual MIDI Input 1 Output out'
	Connected From: 132:0
    5 'Virtual MIDI Output 1 Input in'
	Connecting To: 130:0

air-v, 2022-02-21:
I think the last part of aconnect reveal some weird issue specialy with  client 20, client 134:0 and 134:2

TheKikGen, 2022-02-21:
FYI, issue is corrected for MPC One and Live II.  I'm posting updates img on the github site.
All was linked with the regex pattern that didn"t take ito account different driver names structure.

air-v, 2022-02-21:
GREAT NEWS @TheKikGen.
Did we need to reapply FW update + IAMFORCE or only IAMFORCE Img is enough ?
Thx

TheKikGen, 2022-02-21:
Firmware did'nt change at all since 15 days (and changes were very small things).  
But maybe better to upate again, just to be sure.
You can do that the iamforce sdcard image.   I put these firmwares at the root directory, so you can use the "usb" update mode.

TheKikGen, 2022-02-22:
Updates are online.

air-v, 2022-02-23:
Hello @TheKikGen ,
Sorry but the folder is empty at this time.
I'll retry this afternoon 
Cheers

air-v, 2022-02-23:
Hi,
New files are available, I've tried with tkgl-iamforce-sd-2022-02-22 without success.
Here's some info to help ;

  _____ _          _  ___ _                   _         _
 |_   _| |_  ___  | |/ (_| |____ _ ___ _ _   | |   __ _| |__ ___
   | | | ' \/ -_) | ' <| | / / _` / -_| ' \  | |__/ _` | '_ (_-<
   |_| |_||_\___| |_|\_|_|_\_\__, \___|_||_| |____\__,_|_.__/__/
                             |___/

Welcome to the MPC OS ssh mod !
Linux mpc-one 5.4.124-inmusic-2021-06-10-rt59 #1 SMP PREEMPT_RT Mon Jul 19 17:00:46 UTC 2021 armv7l GNU/Linux
Your IP address is 192.168.0.11
WARNING : you have root permissions. You may harm your system.
root@mpc-one:~# amidi -l
Dir Device    Name
IO  hw:1,0,0  MPC Studio Live Public
IO  hw:1,0,1  MPC Studio Live Private
IO  hw:1,0,2  MPC Studio Live MIDI Port
root@mpc-one:~# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 20: 'MPC One MIDI' [type=kernel,card=1]
    0 'MPC Studio Live Public'
    1 'MPC Studio Live Private'
    2 'MPC Studio Live MIDI Port'
        Connecting To: 134:0
        Connected From: 134:2
client 130: 'Virtual MIDI Output 1 Input' [type=user,pid=289]
    0 'in              '
        Connected From: 134:5
client 131: 'Virtual MIDI Output 2 Input' [type=user,pid=289]
    0 'in              '
        Connected From: 134:3
client 132: 'Virtual MIDI Input 1 Output' [type=user,pid=289]
    0 'out             '
        Connecting To: 134:4
client 133: 'Virtual MIDI Input 2 Output' [type=user,pid=289]
    0 'out             '
        Connecting To: 134:1
client 134: 'MPC' [type=user,pid=260]
    0 'MPC One MIDI MPC Studio Live MIDI Port'
        Connected From: 20:2
    1 'Virtual MIDI Input 2 Output out'
        Connected From: 133:0
    2 'MPC One MIDI MPC Studio Live MIDI Port'
        Connecting To: 20:2
    3 'Virtual MIDI Output 2 Input in'
        Connecting To: 131:0
    4 'Virtual MIDI Input 1 Output out'
        Connected From: 132:0
    5 'Virtual MIDI Output 1 Input in'
        Connecting To: 130:0
root@mpc-one:~#

TheKikGen, 2022-02-23:
Can you post the log here ?
Check in the logs directory : 
/media/TKGL_BOOTSTRAP/tkgl_bootstrap_[ProjectData]/logs

air-v, 2022-02-23:
root@mpc-one:/media/TKGL_BOOTSTRAP/tkgl_bootstrap_[ProjectData]/logs# ls -l
total 4
-rw-r--r--    1 root     root           593 Feb 23 13:01 tkgl_log.txt
root@mpc-one:/media/TKGL_BOOTSTRAP/tkgl_bootstrap_[ProjectData]/logs#

here's properties of Logs files

TheKikGen, 2022-02-23:
Maybe an error in the list of modules to run. Edit this file, 

/media/TKGL_BOOTSTRAP/tkgl_bootstrap_[ProjectData]/doer_list

At the bottom of the file your should have a line : 
TKGL_DOER="playlogo iamforce"

TheKikGen, 2022-02-23:
I changed the batch process to generate images very late in the night....maybe it is not correct.

air-v, 2022-02-23:
# List here your modules name to launch before the MPC application, separated by a space
TKGL_DOER="playlogo"

TKGL_DOER+= iamforce
~
~
~
~
~
~
~

air-v, 2022-02-23:
That was it.... now i'm in FORCE mode 👯

TheKikGen, 2022-02-23:
OK !  Parfait !
You can copy paste the MPC Live definition to the One if you haven't any...
2 sections must exist : 
2/ What buttons-leds in the One
3/ The Mapping rules (src button=dest button)

```
 tklog_info("\n") ;
  tklog_info("--tgkl_help               : Show this help\n") ;
  tklog_info("--tkgl_ctrlname=<name>    : Use external controller containing <name>\n") ;
  tklog_info("--tkgl_iamX               : Emulate MPC X\n") ;
  tklog_info("--tkgl_iamLive            : Emulate MPC Live\n") ;
  tklog_info("--tkgl_iamForce           : Emulate Force\n") ;
  tklog_info("--tkgl_iamOne             : Emulate MPC One\n") ;
  tklog_info("--tkgl_iamLive2           : Emulate MPC Live Mk II\n") ;
  tklog_info("--tkgl_virtualport=<name> : Create end user virtual port <name>\n") ;
  tklog_info("--tkgl_mididump           : Dump original raw midi flow\n") ;
  tklog_info("--tkgl_mididumpPost       : Dump raw midi flow after transformation\n") ;
  tklog_info("--tkgl_configfile=<name>  : Use configuration file <name>\n") ;
  tklog_info("\n") ;
  `exit(0);`
```

air-v, 2022-02-23:
here's an error on  iamforce.log 
[tkgl INFO    ]  --tkgl_configfile specified. File /etc/tkgl_bootstrap/modules/mod_iamforce/conf/map_force.conf will be used for mapping
[tkgl INFO    ]  --tkgl_iamForce specified. Force spoofing.
[tkgl INFO    ]  Original Product code : ACVA (MPC One)
[tkgl INFO    ]  Product code spoofed to ADA2 (Force)
[tkgl INFO    ]  0 keys found in section Map_ONE_FORCE_ButtonsLeds .
[tkgl ***ERROR]  Missing section Map_ONE_FORCE_ButtonsLeds in configuration file /etc/tkgl_bootstrap/modules/mod_iamforce/conf/map_force.conf or syntax error. No mapping set.
[tkgl INFO    ]  MPC controller card id hw:1 found
[tkgl INFO    ]  MPC controller Private port is hw:1,0,1

I will copy MPC Live definition to the One first to check

air-v, 2022-02-23:
Hi,
Here's the leds on MPC ONE  
Sorry @TheKikGen  i don't understand how to use this part of your message 

_3/ The Mapping rules (src button=dest button)

 tklog_info("\n") ;
  tklog_info("--tgkl_help               : Show this help\n") ;
  tklog_info("--tkgl_ctrlname=<name>    : Use external controller containing <name>\n") ;
  tklog_info("--tkgl_iamX               : Emulate MPC X\n") ;
  tklog_info("--tkgl_iamLive            : Emulate MPC Live\n") ;
  tklog_info("--tkgl_iamForce           : Emulate Force\n") ;
  tklog_info("--tkgl_iamOne             : Emulate MPC One\n") ;
  tklog_info("--tkgl_iamLive2           : Emulate MPC Live Mk II\n") ;
  tklog_info("--tkgl_virtualport=<name> : Create end user virtual port <name>\n") ;
  tklog_info("--tkgl_mididump           : Dump original raw midi flow\n") ;
  tklog_info("--tkgl_mididumpPost       : Dump raw midi flow after transformation\n") ;
  tklog_info("--tkgl_configfile=<name>  : Use configuration file <name>\n") ;
  tklog_info("\n") ;
  `exit(0);`_

Thx
[image]

TheKikGen, 2022-02-23:
Concerning the mapping file :

You must defines all buttonsLed .  For example, the Force buttons are defined in the following section :
```
[FORCE_ButtonsLeds]
NAVIGATE=0
KNOBS=1
MENU=2
MATRIX=3
NOTE=4
MASTER=5
CLIP=9
MIXER=11
(...)

```
The declaration format is : (name of the button)=midi value for the button
You can find midi value with amidi in dump mode.  The mpc application must be stopped.
```

[chroot] root@mpc-live:~/MPCLiveXplore-libs/systemctl stop inmusic-mpc

[chroot] root@mpc-live:~/MPCLiveXplore-libs/src# amidi -l
Dir Device    Name
IO  hw:1,0,0  MPC Public
IO  hw:1,0,1  MPC Private
IO  hw:1,0,2  MPC MIDI Port A
IO  hw:1,0,3  MPC MIDI Port B
IO  hw:2,0,0  nanoKEY Studio _ KEYBOARD/CTRL

[chroot] root@mpc-live:~/MPCLiveXplore-libs/src# amidi -h
Usage: amidi options

-h, --help                      this help
-V, --version                   print current version
-l, --list-devices              list all hardware ports
-L, --list-rawmidis             list all RawMIDI definitions
-p, --port=name                 select port by name
-s, --send=file                 send the contents of a (.syx) file
-r, --receive=file              write received data into a file
-S, --send-hex="..."            send hexadecimal bytes
-d, --dump                      print received data as hexadecimal bytes
-t, --timeout=seconds           exits when no data has been received
                                for the specified duration
-a, --active-sensing            include active sensing bytes
-c, --clock                     include clock bytes
-i, --sysex-interval=mseconds   delay in between each SysEx message

[chroot] root@mpc-live:~/MPCLiveXplore-libs/src# amidi -p hw:1,0,1 -d

90 34 7F
90 34 00
```
When you press a button, amidi dumps the corresponding midi message (in hexadecimal).
In the example above, I press the button Main.
```
90 means : button
34 means : code for Main
7F,00 means : pressed / released.

```
If you need to declare this button for the One, you will do the following :
```
[ONE_ButtonsLeds]
MAIN=52   
```
52 beeing the decimal value of 0x34, or :
```
[ONE_ButtonsLeds]
MAIN=0x34   
```
if you prefer hexadecimal value.

Then you need a mapping section beetwin your One and the Force.
Here you set what One button will be mapped to what Force button.
```
##################################################
# ONE to FORCE mapping
##################################################
[Map_ON_FORCE_ButtonsLeds]
# QlinkKnobsShiftMode is used to simulate eight or sixteen qlinks knobs.
# when holding shift, an offset is added to the knob id.
# Authorized value are 0 (disabled) 1 (enabled)
# Note : parameters are always prefixed by "_p_"
_p_QlinkKnobsShiftMode=1

(...)
MAIN=MATRIX
(...)
```

air-v, 2022-02-26:
Hi @TheKikGen ,
I've start to check Button function list to assign and all MPC Live CC are same on ONE too, some buttons are not actually setup and can be used.
First step i need to learn Akai Force  concept (thx youtube) to find what function i can assign.
Second Step: A removable paper or plastic sheet  template will certainly usefull to switch mode to mode .
But mainly first impression is that the KIKPAD with modded firmware is required to using the matrix mode and add some more controllers for live using better easily than the 16 pads on the MPC ONE.
By chance, I've got one, so let's go to flash it and try.

Thx

Potreba, 2022-04-02:
Hello. As of now feb 26 img still has this issue 
TKGL_DOER="playlogo"
TKGL_DOER+= iamforce
Unfortunately editing files within ext4 filesystem proved to be non-trivial for me (win/mac). I suggest copying full repository as zip (maybe I'm too noob for github, who knows?). There are some other issues I have with MPC live mk1 in Iamforce mode. 
1. After 'play start' I cannot go back to regular clip mode. What if I hit this button in a live show, and cannot go back to clips. Not really good
2. How to delete tracks? Seems like some buttons are missing for this function.
3. Is there a way to emulate crossfader? External ctrl would do just fine, but it seems there's no way to map crossfader to external midi controller

I'm very close to perfect live system! Last time I was that excited for sm pro v-machine, which I made run 4 samplelord instances with each part playing its stem.

air-v, 2022-04-04:
Hello Potreba,
1 - Try pressing the NEXT SEQ button and you will return to Launch mode. The PLAY START button is not used to launch the sequence but to switch to chromatic keyboard/drums mode

2 - Currently you cannot delete a track. To delete a track, it would be necessary to have access to the column selection button (which is at the bottom on an Akai Force) and this has not yet been implemented on our MPC.
@TheKikGen is working on Novation Pad mapping for iamForce...and we should definitely have that option. I don't know if the KIKPAD will also be implemented, it could be cool too because this controller has more possibilities than the novation, I think of the potentiometer which could be quite useful).

3 - For the crossfader, I think an external potentiometer can be used, or maybe one of the 4 pots on the MPC will be assignable. Especially if you use the KIKPAD pots.

Hope it help

Potreba, 2022-04-09:
Hi. Thanks for your reply. I'm using LIVE mk1 btw, it makes a bit of difference , because buttons are different. So, here is my .conf file
I've remaped a bunch of buttons, you really have to invest your time into UIing your MPC into FORCE. For example I use +/- as up/down arrows and PLAYSTART as launch 1 scene. These buttons are really close, so I need to push two fingers to launch next scene (I don't think you can do that on force btw), also I've remapped track select and other buttons. Few bugs are still present: sometimes led feedback gets stuck and no TAP blinking, pads not updating colors etc, really annoying. I think its related to 'SHIFT' (maybe), and also 16 pads do not allow for chromatic note mapping (as force/push relies on 5 note spread).
`##################################################
# MPC LIVE to FORCE mapping
##################################################
[Map_LIVE_FORCE_ButtonsLeds]
# QlinkKnobsShiftMode is used to simulate eight or sixteen qlinks knobs.
# when holding shift, an offset is added to the knob id.
# Authorized value are 0 (disabled) 1 (enabled)
# Note : parameters are always prefixed by "_p_"
_p_QlinkKnobsShiftMode=1

# Columns pads
SHIFT_BANK_A=COLUMN_PAD1
SHIFT_BANK_B=COLUMN_PAD2
SHIFT_BANK_C=COLUMN_PAD3
SHIFT_BANK_D=COLUMN_PAD4
BANK_A=MUTE
BANK_B=SOLO
BANK_C=REC_ARM
BANK_D=CLIP_STOP

# Knobs
QLINK1_TOUCH=QLINK1_TOUCH
QLINK2_TOUCH=QLINK2_TOUCH
QLINK3_TOUCH=QLINK3_TOUCH
QLINK4_TOUCH=QLINK4_TOUCH

# Qlink 5 to 8 are simulated with the shift key

QLINK5_TOUCH=QLINK5_TOUCH
QLINK6_TOUCH=QLINK6_TOUCH
QLINK7_TOUCH=QLINK7_TOUCH
QLINK8_TOUCH=QLINK8_TOUCH

# Common buttons
MENU=MENU
PLAY=PLAY
SHIFT_PLAY=NOTE
SHIFT_PLAY_START=LAUNCH
STOP=STOP
REC=REC
MAIN=MATRIX
SHIFT_MAIN=MASTER
NOTE_REPEAT=ARP
COPY=COPY
ERASE=DELETE
TAP_TEMPO=TAP_TEMPO
SHIFT_TAP_TEMPO=SHIFT_MUTE_PAD5
FULL_LEVEL=SELECT
SHIFT_FULL_LEVEL=EDIT
16_LEVEL=CLIP
SHIFT_16_LEVEL=MIXER
UNDO=UNDO
# REDO
SHIFT_UNDO=SHIFT_UNDO
SHIFT_STOP=STOP_ALL
PLAY_START=LAUNCH_1
SHIFT_MENU=LOAD
OVERDUB=SAVE
PLUS=UP
MINUS=DOWN
QLINK_SELECT=KNOBS
SHIFT_PLUS=RIGHT
SHIFT_MINUS=LEFT`

closed 2023-03-31 by TheKikGen