TheKikGen/MPC-LiveXplore #34: [iamForce] Shutting down button or New Project failed and go back to MPC with internal error message

air-v, 2022-02-27:
Hi,
When i want to shutting down.iamForce or create new project . iamForce go back on normal MPC soft but with message " Your MPC recovered from an internal error" .
Maybe it's normal ?

TheKikGen, 2022-03-11:
Not really normal, but this is an issue I have when the Force software runs on the MPC hardware.
Due to this issue, I changed the chroot approach to a full "mount bind" approach, in the same spirit however : protecting the host.  
It is not possible to shuwdown within a chroot because dbus is not started...

air-v, 2022-03-11:
Thanks for clarification,
I understand that it can't be stopped as usual way and that's okay for me. The problem with "new project" option is the most annoying.