TheKikGen/MPC-LiveXplore #37: I am trying to adapt it for Denon Prime4...

dzelionis, 2022-04-23:
Hey, I have managed to do good few succesfull custom firmaware updates on Denon Prime4...:D
Still having issues getting sshd to work, as on thelates firmware version that was wiped from firmware completly....
Also i am not sure on the console (debug) as  i receiving data, but i can not enter anything so, my question is on U-boot:
[image]

does gpio-keys for in means that i have no serial input and i need something special or how it is...as i could get working other direction......
Just for reference boot log from prime4 (factory firmware):

U-Boot SPL 2016.05-inmusic-20190128 (May 21 2019 - 16:13:39)
Trying to boot from MMC1

U-Boot 2016.05-inmusic-20190128 (May 21 2019 - 16:13:39 +0000), Build: jenkins-AIRDJ-Embedded-Release-162

Model: Denon DJ Prime 4
DRAM:  2 GiB
MMC:   dwmmc@ff0f0000: 0
In:    gpio-keys
Out:   serial
Err:   serial
AZ01 PCB revD (raw=13)
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:6...
Found U-Boot script /boot/boot.scr.uimg
489 bytes read in 8 ms (59.6 KiB/s)
## Executing script at 00000000
4792072 bytes read in 222 ms (20.6 MiB/s)
51714 bytes read in 12 ms (4.1 MiB/s)
Kernel image @ 0x2000000 [ 0x000000 - 0x491f08 ]
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Device Tree to 1fff0000, end 1ffffa01 ... OK

Starting kernel ...

000: Booting Linux on physical CPU 0x500
000: Linux version 5.4.131-inmusic-2021-07-13-rt61 (buildroot@buildroot) (gcc version 10.3.0 (Buildroot 2021.02.3-12-g23d424b4ea)) #1 SMP PREEMPT_RT Wed Feb 9 14:21:02 UTC 2022
000: CPU: ARMv7 Processor [410fc0d1] revision 1 (ARMv7), cr=10c5387d
000: CPU: div instructions available: patching division code
000: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
000: OF: fdt: Machine model: Denon DJ Prime 4
000: Memory policy: Data cache writealloc
000: OF: reserved mem: initialized node framebuffer, compatible id rockchip,framebuffer-memory
000: percpu: Embedded 12 pages/cpu s16864 r8192 d24096 u49152
000: Built 1 zonelists, mobility grouping on.  Total pages: 522752
000: Kernel command line: root=PARTUUID=24d1deac-3434-1a4e-98d1-68ee2945a5f1 rootwait ro rfkill.default_state=0  isolcpus=1-3
000: Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
000: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
000: mem auto-init: stack:off, heap alloc:off, heap free:off
000: Memory: 2054828K/2097152K available (7168K kernel code, 405K rwdata, 2192K rodata, 1024K init, 267K bss, 30036K reserved, 12288K cma-reserved, 1298432K highmem)
000: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
000: rcu: Preemptible hierarchical RCU implementation.
000: rcu:       RCU priority boosting: priority 1 delay 500 ms.
000: rcu:       RCU_SOFTIRQ processing moved to rcuc kthreads.
000:    No expedited grace period (rcu_normal_after_boot).
000:    Tasks RCU enabled.
000: rcu: RCU calculated value of scheduler-enlistment delay is 100 jiffies.
000: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
000: random: get_random_bytes called from start_kernel+0x264/0x45c with crng_init=0
000: rockchip: reset MMC AHB arbitor
000: arch_timer: cp15 timer(s) running at 24.00MHz (phys).
000: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
000: sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
000: Switching to timer-based delay loop, resolution 41ns
000: Console: colour dummy device 80x30
000: Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=24000)
000: pid_max: default: 32768 minimum: 301
000: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
000: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[... 462 lines cut ...]
[  OK  ] Reached target Login Prompts.
         Starting WPA supplicant...
         Starting Firmware update for XMOS USB audio devices...
[  OK  ] Finished Remove all but the most recent core dump.
         Starting Hostname Service...
[  OK  ] Started Avahi mDNS/DNS-SD Stack.
[  OK  ] Started Bluetooth service.
[  OK  ] Reached target Bluetooth.
[FAILED] Failed to start Firmware u…te for XMOS USB audio devices.
See 'systemctl status xmos-update.service' for details.
         Starting Engine...
[  OK  ] Started WPA supplicant.
[  OK  ] Finished Network Connectivity.
[  OK  ] Started Engine.
[  OK  ] Started Hostname Service.
[  OK  ] Finished Hostname setup for AZ01.
[  OK  ] Reached target Network (Pre).
         Starting Connection service...
[  OK  ] Started Connection service.
[  OK  ] Reached target Network.
[  OK  ] Started AZ01 script runner service.
[  OK  ] Reached target Multi-User System.

Welcome to Buildroot
prime4 login:

TheKikGen, 2022-04-24:
Hi.  Mount the image on a Linux computer (uncompress it with the tools you will find on this site).
Modify the launch script in /bin and maybe try to launch telnetd from Busybox.
Rebuild the image with the img tool and update again.  You should get a telnet access.
Im quite suprised InMusic deleted ssh. Are you sure ? Check systemctl services .

TheKikGen, 2022-04-24:
https://github.com/TheKikGen/MPC-LiveXplore/tree/master/imgmaker

dzelionis, 2022-04-24:
> Hi. Mount the image on a Linux computer (uncompress it with the tools you will find on this site). Modify the launch script in /bin and maybe try to launch telnetd from Busybox. Rebuild the image [...]

i can be sure yet, but its no sign off ssh in /etc/ or /bin/ or /usr/sbin/ unless its in overlay somewher in overlay which i don't see yet, as i strugling with concole input, but its definetly not in the firmware update for sure.... i am linux user ( in real life i am senior network engineer, using linux desktop for day to day work).....yes u have succesfully adapted  your python/shell scripts to match my usb vids / devices ids...etc... and i manage to do ~3 succesfull firmware upgrades playing with different modifications.....so i stiill thinking how i could get console to to work...there is file in "/bin/az01-stup-otgconsole.sh" which is looking interesting for but i cannot figure out is on some other usb port which is hidden or something  as i dont see dummy id's on linux box (have tried all the different modes on prime: normal/fastboot/pc) .. so if i could figure out this one that also would help for digging in inside device filesystem...

just for reference file cotents:
#!/bin/sh
#
# This script follows the guide in Documentation/usb/gadget_configfs.txt
#
# You can access the other end of this on a Linux system by binding the
# VID/PID used below to the generic USB-serial driver:
#
#       echo '0001 0001' | sudo tee /sys/bus/usb-serial/drivers/generic/new_id

# systemd automounts configfs at /sys/kernel/config so no need to mount here.
(
    cd /sys/kernel/config &&
    ### Basic device setup:
    mkdir -p usb_gadget/g1 &&
    cd usb_gadget/g1 &&
    # FIXME: dummy VID & PID
    echo 0x0001 >idVendor &&
    echo 0x0001 >idProduct &&
    # note: full list of possible parameters in
    #       Documentation/ABI/testing/configfs-usb-gadget
    mkdir -p strings/0x409 &&
    # FIXME: dummy values below
    echo 1234567890 >strings/0x409/serialnumber &&
    echo 'Dummy Corp.' >strings/0x409/manufacturer &&
    echo 'Dummy Product' >strings/0x409/product &&
    ### Configuration:
    mkdir -p configs/c.1 &&
    (
        cd configs/c.1 &&
        mkdir -p strings/0x409 &&
        echo 'My Configuration' >strings/0x409/configuration &&
        echo 100 >MaxPower
    ) &&
    ### Functions:
    mkdir functions/gser.otg0 &&
    ### Associate functions with configurations:
    ln -s functions/gser.otg0 configs/c.1 &&
    ### Enable the gadget:
    echo ff580000.usb >UDC &&
    sleep 1 &&
    systemctl start serial-getty@ttyGS"$(cat functions/gser.otg0/port_num)"
)

Thanks
Best regards
Domas Zelionis

dzelionis, 2022-04-25:
[image]

dzelionis, 2022-04-25:
[image]
[image]
[image]
[image]
[image]

Modification on the motherboard you see came from factory....does not look great....

TheKikGen, 2022-06-30:
A nice page to add to the WIKI

ounsatn, 2024-04-26:
Hello did you succeed to connect to your prime4 ? 
in service man of the akf its like this, i think you have some pins of the cpu fan, which are for the TX. 
[image]

osab83, 2024-07-15:
Hello sirs, how long did it take you to put firmware back into the Denon Prime 4?

osab83, 2024-07-15:
I also have a Denon Prime 4, and I think the firmware comes in two parts, one for the mixer and another for the screen or am I wrong? In my case the firmware that manages the mixer does not work correctly, it does not start and it does not allow me to update the mixer firmware

ounsatn, 2024-07-15:
Hello , the mainboard & cpu is driven by a custom linux. 
Some internal components as STM32F have firmware and are updated via dfu interface.
In assignment files for Engine on P4 you have 3 qml : Controller, Display & Mixer. I suppose you have 3 interfaces when you do amidi -l . 
have a look on dfu programming , there are tools like dfu-util which allow to dump & restore content of the chip.

osab83, 2024-07-15:
[
[image]
](url)

ounsatn, 2024-07-15:
you need console access to see what happens really, this screen is only a frontend for users. 
either you find a firmware with ssh or telnet activated, or either you plug TTL/USB converter on the console port

osab83, 2024-07-15:
Look, the strange thing about this is that we wrote again to the eeprom, and the problem continued, the ima is already mounted again, we are in doubt, what to modify to get the firmware up again

ounsatn, 2024-07-15:
what is engine saying when running it with /usr/engine/Engine -d0  ?

osab83, 2024-07-15:
What are you referring to? /usr/engine/Engine -d0 ?

I should run it again and see what I code, command..
Which firmware version did you use without problems?

ounsatn, 2024-07-15:
i dont have p4, i have a prime-go and an akai force.
I use firmware 4.0 on pgo. 
 
When system boots, it runs engine.service via systemctl. 
This service launch /usr/engine/Engine . 

if you want to stop it , do systemctl stop engine.service, then run it by hand with command : 

/usr/engine/Engine -d0 

d0 : is for debug message goes to console ( otherwise it goes to a logfile in /media/az01-internal/AppDataUser/AIR Music Technology/Engine/Logs ) 

sometimes you'll have to setup env var below : 

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/qt/lib
export AIR_SCREEN_ROTATION=-90

you can do this when having a console access to the embedded linux ( via ssh or via serial port)

osab83, 2024-07-15:
I did it working on Windows, could that be the error? should I use linux?

ounsatn, 2024-07-15:
no, i'm talking about the linux embedded on prime4. if you want to understand what happens you have to get debug output. see first post from [dzelionis]. your p4 is no more under warrantly ?

osab83, 2024-07-15:
Apparently without guarantee, Denon no longer responds to the messages.. well we will see what to do so that the machine does not become obsolete

dzelionis, 2024-07-17:
@osab83 , if you looking for ssh enabled firmware with root access, you can get it from https://github.com/icedream/denon-prime4 . After you will get access to root shell, you can check boot logs by running "dmesg". That may give you some clues whats wrong with the device.... In case, something goes wrong with flashing and you brick your device (very low risk), you always have an option to press a button on the motherboard to put it into Fastboot mode and push new software again... I hope that's helps...

osab83, 2024-07-17:
Hello, do you think I can put a new eeprom and write again? It would solve the problem, you would have the data on what type of eeprom would be indicated

Obtener Outlook para Android<https://aka.ms/ghei36>

dzelionis, 2024-07-18:
> Hello, do you think I can put a new eeprom and write again? It would solve the problem, you would have the data on what type of eeprom would be indicated Obtener Outlook para Android<https://aka.ms/ghei36>
> […](#)

I am not sure. First, could you explain how did it failed ? Did it failed when you was doing software upgrade? or it just failed while using device ? In case it failed while doing firmware upgrade, I think you have good chances to fix it by doing another attempt of software downgrade/upgrade as software upgrade image will update all the eeprom's and in case of eeprom flashing process got interrupted / corrupted while doing software upgrade - 2nd attempt should solve that...

Would be different case if stuff has failed while using device, in that case probably there an issue with data channel between motherboard and mixerboard...

dzelionis, 2024-07-18:
in case you cannot put device in software upgrade mode using screen, you can push software using "fastboot" utility ( https://source.android.com/docs/setup/test/running ) , you just need to put your motherboard to fastboot mode and you can do that by pressing and holding "Recovery" button on motherboard while powering device.  
[image]

Some info how to upgrade software using fastboot you can find here:
https://github.com/icedream/denon-prime4/issues/15

you just need to extract rootfs image from denon firmware upgrade if you using this method.

also you may check all the I2S cables/connectors (if its fully seeded or just reseed it) between motherboard and mixerboard as i think it would use these for programming eeprom on mixerboard

osab83, 2024-07-18:
> > Hola, crees que pueda poner una eeprom nueva y volver a escribir? Solucionaría el problema, tendrías el dato de que tipo de eeprom sería el indicado Obtener Outlook para Android [https://aka.ms/ghei36 ](https://aka.ms/ghei36)
> > […](#) 
> 
> No estoy seguro. Primero, ¿podrías explicarme cómo falló? ¿Falló cuando estabas haciendo la actualización de software? ¿O simplemente falló mientras usabas el dispositivo? En caso de que fallara mientras estabas haciendo la actualización de firmware, creo que tienes buenas posibilidades de solucionarlo haciendo otro intento de actualización/degradación de software, ya que la imagen de actualización de software actualizará todas las EEPROM y, en caso de que el proceso de actualización de la EEPROM se interrumpiera o se corrompiera mientras se hacía la actualización de software, el segundo intento debería resolverlo...
> 
> Sería un caso diferente si algo fallara mientras se usaba el dispositivo, en ese caso probablemente haya un problema con el canal de datos entre la placa base y la mesa de mezclas...

It fails in operation, when I turn the machine back on, I notice that my peripherals are from the mixer, and it sends me a mixer update SMS, which it cannot perform, it tries to update, but it does not do it,

dzelionis, 2024-07-18:
@osab83, in that case, if I was you, i would check/re-seed all the cables in between motherboard and mixerboard...
also, you may check your USB ports on the devices, see if there any damaged usb port, that could also cause  some issues if some pin usb port touching ground....

osab83, 2024-07-18:
> @osab83, en ese caso, si yo fuera tú, revisaría/volvería a conectar todos los cables entre la placa base y la mesa de mezclas... además, puedes revisar los puertos USB en los dispositivos, ver si hay algún puerto USB dañado, eso también podría causar algunos problemas si algún pin del puerto USB toca tierra...

ok friends