TheKikGen/MPC-LiveXplore #40: MPC 2.11.1 update

IrregularShed, 2022-06-28:
Hi everyone

The new OS is out and I'd like to have it SSH enabled so that I can do file handling on my PC, but having opened the update.img inside the update application I've realised out of my depth I am!

Is there likely to be a pre-patched image in the near future?

Thanks for everything!

TheKikGen, 2022-06-30:
I will upload a new patched img on next saturday.  Image already done and patched, ready to be uploaded.

LoopinLab, 2022-10-10:
Unfortunately, AKAI has introduced a nasty bug in 2.11.2 that stops the Looper whenever you change to another view (such as the Track Mute Screen). It seems they fixed it in 2.11.4 

This pretty much ruins our Live performance. Before moving back to 2.10.1 - can you tell me if there's a chance that you develop an SSH version of 2.11.4 or 2.11.5 anytime soon? That would be awesome. 

All the Best, Michael.

TheKikGen, 2022-10-15:
Yes.  It is exactly the same for the last 2.11.5 with MP3.  Not usable anymore.  I'll wait for a  more stable version.
2.11.6 ???

LoopinLab, 2022-10-26:
2.11.6 is out. But no release notes about bugfixes. 🙄

TheKikGen, 2022-11-11:
2.11.6 patched with SSH mod and available on the home page here...

LoopinLab, 2022-11-11:
Awesome! I already installed it and it looks good. The Looper works again! Thank you very much!

closed 2023-03-31 by TheKikGen