TheKikGen/MPC-LiveXplore #43: How to use Mod_Anyctrl ?

Potreba, 2022-11-16:
Hi! I'm looking for ways to control IamForce with external midi controller. I'm looking for info to build Arduino / Opendeck midi controller as a footwsitch to do stuff like up/down arrow , combined down/launch_scene8, maybe expression pedal to crossfader. 
As far as I understand Mod_anyctrl is a module responsible for such task? Is there a way to map midi CC / note like you did with mpc.conf file?

TheKikGen, 2023-01-03:
**mod_anyctrl** will insert your controller in the internal midi loop of your Force.
If you send the correct midi message from your controller, you can simulate any button of the Force.
Basically, if your want to PLAY when your press your foot controller,  it must send the PLAY midi message.
You can hardcode that rapidly in your Arduino source.
I also did midi usb interfaces with Arduino and STM32F103 uC.  [Usbmidiklik](https://github.com/TheKikGen/USBMidiKliK).

You have to pass the usb name (use "lsusb" on the Linux command line) of your controller as a command line parameter to mod_anyctrl so it will recognize it.

TheKikGen, 2023-01-03:
Midi controls for MPC and Force (draft document) :
https://docs.google.com/spreadsheets/d/1FIs0A5jHfwl3VmDx-JLCFPhBvkAvlwbQu6nkoF2CoRg/edit#gid=692652057

```
b0 00 00-7f - Navigate
b0 01 00-7f - Knobs
b0 02 00-7f - Menu
b0 03 00-7f - Matrix
b0 04 00-7f - Note
b0 05 00-7f - Master
b0 09 00-7f - Clip
b0 0b 00-7f - Mixer
b0 23 00-7f - Load
b0 24 00-7f - Save
b0 25 00-7f - Edit
b0 26 00-7f - Delete
b0 31 00-7f - Shift
b0 34 00-7f - Select
b0 35 00-7f - Tap Tempo
b0 36 00-7f - +
b0 37 00-7f - -
b0 38 00-7f - Launch 1
b0 39 00-7f - Launch 2
b0 3a 00-7f - Launch 3
b0 3b 00-7f - Launch 4
b0 3c 00-7f - Launch 5
b0 3d 00-7f - Launch 6
b0 3e 00-7f - Launch 7
b0 3f 00-7f - Launch 8
b0 43 00-7f - Undo
b0 49 00-7f - Rec
b0 51 00-7f - Stop
b0 52 00-7f - Play
b0 5b 00-7f - Mute
b0 5c 00-7f - Solo
b0 5d 00-7f - Rec Arm
b0 5e 00-7f - Clip Stop
b0 5f 00-7f - Stop All
b0 70 00-7f - ^
b0 71 00-7f - v
b0 72 00-7f - <
b0 73 00-7f - >
b0 74 00-7f - Launch
b0 75 00-7f - Step Seq
b0 76 00-7f - Arp
b0 7a 00-7f - Copy
b0 7b 00-7f - Assign A
b0 7c 00-7f - Assign B

```

TheKikGen, 2023-01-03:
# export ANYCTRL_NAME="Launchpad Pro MK3";LD_PRELOAD=/media/MSTICK/tkgl_anyctrl.so /usr/bin/FORCE

TheKikGen, 2023-01-06:
Yes  ANYCTRL_NAME="my usb controller name".

00-7F is the range in deed.  That means the third byte must be in that range

air-v, 2023-01-09:
Hello, First of all HAPPY NEW YEAR. Wish you the best
I'm trying to launch this with KIKPAD connected on it (the device name after LSUSB)
#  export ANYCTRL_NAME="Focusrite-Novation Launchpad Mini MK3";LD_PRELOAD=/media/HAKAI/tkgl_anyctrl.so /usr/bin/FORCE
------------------------------------
TKGL_ANYCTRL V1.0 by the KikGen Labs
------------------------------------
(tkgl_anyctrl) **** Error : MPC card not found
I've got error. what do i need to add ?
Regards
Herve

TheKikGen, 2023-01-09:
Thanks !  Happy new year to uou too.
Can you post here the result of lsusb command when the Kikpad is plugged in ?
(I suppose you are using the Novation emulation firmware ?)
Try with "Launchpad Mini MK3" only.

air-v, 2023-01-09:
Hi,Here's the result without success 

# lsusbBus 001 Device 002: ID 09e8:0046 AKAI  Professional M.I. Corp. MPC One MIDIBus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hubBus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hubBus 002 Device 004: ID 1235:0113 Focusrite-Novation Launchpad Mini MK3Bus 002 Device 005: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridgeBus 002 Device 003: ID 045b:0209 Hitachi, LtdBus 002 Device 002: ID 045b:0209 Hitachi, LtdBus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub#  export ANYCTRL_NAME="Focusrite-Novation Launchpad Mini MK3";LD_PRELOAD=/media/HAKAI/tkgl_anyctrl.so /usr/bin/FORCE------------------------------------TKGL_ANYCTRL V1.0 by the KikGen Labs------------------------------------(tkgl_anyctrl) **** Error : MPC card not found#  export ANYCTRL_NAME="Launchpad Mini MK3";LD_PRELOAD=/media/HAKAI/tkgl_anyctrl.so /usr/bin/FORCE------------------------------------TKGL_ANYCTRL V1.0 by the KikGen Labs------------------------------------(tkgl_anyctrl) **** Error : MPC card not found#  export ANYCTRL_NAME="Novation Launchpad Mini MK3";LD_PRELOAD=/media/HAKAI/tkgl_anyctrl.so /usr/bin/FORCE------------------------------------TKGL_ANYCTRL V1.0 by the KikGen Labs------------------------------------(tkgl_anyctrl) **** Error : MPC card not found#
Regards

Herve Laporte

TheKikGen, 2023-01-09:
Can your post result of "aconnect -l " ?

air-v, 2023-01-09:
And here it is
***@***.***:/media/TKGL_BOOTSTRAP# aconnect -lclient 0: 'System' [type=kernel]    0 'Timer           '    1 'Announce        'client 16: 'MPC One MIDI' [type=kernel,card=0]    0 'MPC Studio Live Public'        Connected From: 136:0    1 'MPC Studio Live Private'        Connecting To: 134:0        Connected From: 135:0    2 'MPC Studio Live MIDI Port'        Connecting To: 137:0        Connected From: 137:2client 20: 'Launchpad Mini MK3' [type=kernel,card=1]    0 'KIKPAD IN       '        Connecting To: 134:0        Connected From: 135:0, 136:0    1 'KIKPAD IN       'client 130: 'Virtual MIDI Output 1 Input' [type=user,pid=279]    0 'in              '        Connected From: 137:5client 131: 'Virtual MIDI Output 2 Input' [type=user,pid=279]    0 'in              '        Connected From: 137:3client 132: 'Virtual MIDI Input 1 Output' [type=user,pid=279]    0 'out             '        Connecting To: 137:4client 133: 'Virtual MIDI Input 2 Output' [type=user,pid=279]    0 'out             '        Connecting To: 137:1client 134: 'Client-134' [type=user,pid=333]    0 'TKGL Virtual In Private'        Connected From: 16:1, 20:0client 135: 'Client-135' [type=user,pid=333]    0 'TKGL Virtual Out Private'        Connecting To: 16:1, 20:0client 136: 'Client-136' [type=user,pid=333]    0 'TKGL Virtual Out Public'        Connecting To: 16:0, 20:0client 137: 'MPC' [type=user,pid=333]    0 'MPC One MIDI MPC Studio Live MIDI Port'        Connected From: 16:2    1 'Virtual MIDI Input 2 Output out'        Connected From: 133:0    2 'MPC One MIDI MPC Studio Live MIDI Port'        Connecting To: 16:2    3 'Virtual MIDI Output 2 Input in'        Connecting To: 131:0    4 'Virtual MIDI Input 1 Output out'        Connected From: 132:0    5 'Virtual MIDI Output 1 Input in'        Connecting To: 130:0
Herve Laporte

air-v, 2023-01-09:
Here it is :
root@mpc-one:/media/TKGL_BOOTSTRAP# aconnect -l
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 16: 'MPC One MIDI' [type=kernel,card=0]
    0 'MPC Studio Live Public'
        Connected From: 136:0
    1 'MPC Studio Live Private'
        Connecting To: 134:0
        Connected From: 135:0
    2 'MPC Studio Live MIDI Port'
        Connecting To: 137:0
        Connected From: 137:2
client 20: 'Launchpad Mini MK3' [type=kernel,card=1]
    0 'KIKPAD IN       '
        Connecting To: 134:0
        Connected From: 135:0, 136:0
    1 'KIKPAD IN       '
client 130: 'Virtual MIDI Output 1 Input' [type=user,pid=279]
    0 'in              '
        Connected From: 137:5
client 131: 'Virtual MIDI Output 2 Input' [type=user,pid=279]
    0 'in              '
        Connected From: 137:3
client 132: 'Virtual MIDI Input 1 Output' [type=user,pid=279]
    0 'out             '
        Connecting To: 137:4
client 133: 'Virtual MIDI Input 2 Output' [type=user,pid=279]
    0 'out             '
        Connecting To: 137:1
client 134: 'Client-134' [type=user,pid=333]
    0 'TKGL Virtual In Private'
        Connected From: 16:1, 20:0
client 135: 'Client-135' [type=user,pid=333]
    0 'TKGL Virtual Out Private'
        Connecting To: 16:1, 20:0
client 136: 'Client-136' [type=user,pid=333]
    0 'TKGL Virtual Out Public'
        Connecting To: 16:0, 20:0
client 137: 'MPC' [type=user,pid=333]
    0 'MPC One MIDI MPC Studio Live MIDI Port'
        Connected From: 16:2
    1 'Virtual MIDI Input 2 Output out'
        Connected From: 133:0
    2 'MPC One MIDI MPC Studio Live MIDI Port'
        Connecting To: 16:2
    3 'Virtual MIDI Output 2 Input in'
        Connecting To: 131:0
    4 'Virtual MIDI Input 1 Output out'
        Connected From: 132:0
    5 'Virtual MIDI Output 1 Input in'
        Connecting To: 130:0
root@mpc-one:/media/TKGL_BOOTSTRAP#

TheKikGen, 2023-01-09:
If you are using the bootstrap image,  the Kikpad with the Launchpad MK3 emulation should work.

TheKikGen, 2023-01-09:
check in the module script tkgl_mod_iamforce.sh

TheKikGen, 2023-01-09:
Alternatively to tkgl_anyctrl.so, you can use mpcmapper.so.  
Ex : 
```
LD_PRELOAD=./tkgl_mpcmapper.so ../../MPCF-306  --tkgl_iamForce --tkgl_configfile=tkgl_mpc
mapper.ini --tkgl_ctrlname="Launchpad Mini.*(KIKPAD|MIDI).*"
```

TheKikGen, 2023-01-09:
Or withut config file : 

LD_PRELOAD=./tkgl_mpcmapper.so ../../MPCF-306  --tkgl_ctrlname="Launchpad Mini.*(KIKPAD|MIDI).*"

Potreba, 2023-01-10:
@TheKikGen happy new year to you!

I've added anyctrl to doer_list . If I add it first, iamforce never starts (plays kikgen logo and back to mpc). I've tried editing tkgl_mod_anyctrl.sh for my test controller. I'm not a programmer, so I've copied your code into every place I think it might've work. But it didn't . I tried sending 'Play" command and it is b0 52 00-0F = 1 chan CC 82 00-127 . Nothing happens 

#!/bin/sh
#
# __ __| |           |  /_) |     ___|             |           |
#    |   __ \   _ \  ' /  | |  / |      _ \ __ \   |      _` | __ \   __|
#    |   | | |  __/  . \  |   <  |   |  __/ |   |  |     (   | |   |\__ \
#   _|  _| |_|\___| _|\_\_|_|\_\\____|\___|_|  _| _____|\__,_|_.__/ ____/
#
# BOOTSTRAP script for ANYCTRL LD_PRELOAD LIBRARY.

SCRIPT_NAME=$(readlink -f "$0")
SCRIPT_DIR=$(dirname "$SCRIPT_NAME")
source "$SCRIPT_DIR/../../scripts/tkgl_path"

# This "low-level" library allows you to set up any controller as a control
# surface to drive the MPC standalone application (FULL PRIVATE AND PUBLIC PORTS).
#------------------------------------------------------------------------------

# Enter here the name (or a substring)  of your external midi controller.
# You can use the command 'amidi -l' to find it

export ANYCTRL_NAME="ReMOTE SL MIDI 1"
export ANYCTRL_NAME="ReMOTE SL MIDI 2"
export ANYCTRL_NAME="ReMOTE SL MIDI 3"
export ANYCTRL_NAME="Focusrite-Novation RemoteSL + ZeroSL"

#------------------------------------------------------------------------------

# Launch MPC application as usual, but take command line difference in account
# The Force has one.., MPCs haven't. Don't know if this is important.
ARGV=""
[ "$DEVICE" == "Force" ] && ARGV="$@"

#------------------ original script start here---------------------------------
# find dfu util info
DFUUTILINFO=$(dfu-util -l | grep "0x08000000")

#if in dfu-update mode run firmware update script
[ "x$DFUUTILINFO" != "x" ] && sh /usr/share/Akai/SME0/Firmware/update.sh

export ANYCTRL_NAME="ReMOTE SL MIDI 1";LD_PRELOAD=$TKGL_LIB/tkgl_anyctrl.so
export ANYCTRL_NAME="Focusrite-Novation RemoteSL + ZeroSL";LD_PRELOAD=$TKGL_LIB/tkgl_anyctrl.so

if type systemd-inhibit >/dev/null 2>&1
then
    exec systemd-inhibit --what=handle-power-key /usr/bin/MPC $ARGV
else
    # Reduce the soft stack size limit to 1MiB (from the default 8MiB).  This
    # is used as the default stack size for new threads and since MPC is
    # locked in memory this memory is allocated.  Since the limit is read
    # before main, we cannot set it in the application and must do so here.
    ulimit -S -s 1024
    exec /usr/bin/MPC $ARGV
fi

TheKikGen, 2023-01-10:
I just pushed a correction tkgl_anyctrl.c
You fill find the new library so file here : https://github.com/TheKikGen/MPCLiveXplore-libs/tree/main/src
`export ANYCTRL_NAME="Launchpad Mini.*(KIKPAD|MIDI).*";LD_PRELOAD=./tkgl_anyctrl.so /usr/bin/MPC`

TheKikGen, 2023-01-10:
@Potreba  you script can't work as is.

post here the result of a "aconnect -l" command.

TheKikGen, 2023-01-11:
What MPC do you own ?

It seems the pattern to detect the MPC One is not working correctly in the source code of anyctrl.
I will correct that later.

As said, mpcmapper can be used. It is a more advanced "anyctrl".
Check the libs repository : https://github.com/TheKikGen/MPCLiveXplore-libs/tree/main/src

Maybe preferable to test with the IamForce sdcard image . download and flash on a sdcard : 
https://drive.google.com/drive/u/5/folders/16tqhdznLIwpyl2uw8BjkqHjZw-EEpEF1

if IamForce starts, that means everything is ok.

TheKikGen, 2023-01-12:
@Potreba 
Ok. You must copy the last version of tkgl_anyctrl.so mentioned above to the tkgl_bootstrap_[ProjectData]/lib/ directory first.
Then, you can use that script version with your Remote SL controller :
```
#!/bin/sh
#
# __ __| |           |  /_) |     ___|             |           |
#    |   __ \   _ \  ' /  | |  / |      _ \ __ \   |      _` | __ \   __|
#    |   | | |  __/  . \  |   <  |   |  __/ |   |  |     (   | |   |\__ \
#   _|  _| |_|\___| _|\_\_|_|\_\\____|\___|_|  _| _____|\__,_|_.__/ ____/
#
# BOOTSTRAP script for ANYCTRL LD_PRELOAD LIBRARY.

SCRIPT_NAME=$(readlink -f "$0")
SCRIPT_DIR=$(dirname "$SCRIPT_NAME")
source "$SCRIPT_DIR/../../scripts/tkgl_path"

# This "low-level" library allows you to set up any controller as a control
# surface to drive the MPC standalone application (FULL PRIVATE AND PUBLIC PORTS).
#------------------------------------------------------------------------------

# Enter here the name (or regex pattern)  of your external midi controller.
# You can use the command 'amidi -l' to find it
# Example : ANYCTRL_NAME="Launchpad Mini.*(KIKPAD|MIDI).*"

# Capture the Remote SL first port with a regex pattern
# client 28: 'ReMOTE SL' [type=kernel,card=3]
#    0 'ReMOTE SL MIDI 1'

export ANYCTRL_NAME="ReMOTE SL.*"

#------------------------------------------------------------------------------

# Launch MPC application as usual, but take command line difference in account
# The Force has one.., MPCs haven't. Don't know if this is important.
ARGV=""
[ "$DEVICE" == "Force" ] && ARGV="$@"

#------------------ original script start here---------------------------------
# find dfu util info
DFUUTILINFO=$(dfu-util -l | grep "0x08000000")

#if in dfu-update mode run firmware update script
[ "x$DFUUTILINFO" != "x" ] && sh /usr/share/Akai/SME0/Firmware/update.sh

export LD_PRELOAD=$TKGL_LIB/tkgl_anyctrl.so

if type systemd-inhibit >/dev/null 2>&1
then
    exec systemd-inhibit --what=handle-power-key /usr/bin/MPC $ARGV
else
    # Reduce the soft stack size limit to 1MiB (from the default 8MiB).  This
    # is used as the default stack size for new threads and since MPC is
    # locked in memory this memory is allocated.  Since the limit is read
    # before main, we cannot set it in the application and must do so here.
    ulimit -S -s 1024
    exec /usr/bin/MPC $ARGV
fi

```

Potreba, 2023-01-12:
I've copied files from your link above to 'lib' folder and copied code you've written above to tkgl_mod_anyctrl.sh . Still nothing works. I'm sending midi messages b0 52 00-7f - Play from SL . This is hex, right, so its CC#82.

TheKikGen, 2023-01-13:
Connect with ssh :  
`# ssh root@<your mpc ip address>
`   
Stop the MPC application  : 
`#  systemctl stop inmusic-mpc`

try manually to launch MPC with anyctrl (with the bootstrap sdcard in) :

```
# cd /media/TKGL_BOOTSTRAP/tkgl_bootstrap_[ProjectData]/lib
# export ANYCTRL_NAME="ReMOTE SL.*";LD_PRELOAD=./tkgl_anyctrl_lt.so /usr/bin/MPC
```

Post the result here

closed 2023-03-31 by TheKikGen