TheKikGen/MPC-LiveXplore #49: No boot on SD card....

air-v, 2023-03-31:
Hello Fitz,
I'm trying to use SD card to boot on iamforce2  and it's not working. Any Idea?
Regards

TheKikGen, 2023-04-01:
I need a bit more....

Do you see the logo ?
What is the doer_list content ?

What' in the log at tkgl_bootstrap_[ProjectData]/logs/  ?

air-v, 2023-04-01:
Yes it's a MPC One with FW 2.11.8tkglssh installed
No KIKGEN Logo appears  ...look like it doesn't boot on SD card reader
In DOER file, I use this option : TKGL_DOER="playlogo iamforce2" because the pad option @KIKPADMK3 doesn't work for me (even with USB KEY), so i add it in .sh file
There's nothing inside Logs folder

realitygaps, 2023-04-03:
i also am having no luck running iamforce2 on sdcard on mpc one. It seems tkgl_path is also missing from these images, which is called by the modules but not in the scripts directory where expected

realitygaps, 2023-04-06:
@TheKikGen is there maybe an example/copy of tkgl_path anywhere so I can try fix the errors i see running modules?

mabeltron, 2023-05-07:
I have a similar issue with the current image on SD card on MPC Live II. It's showing the Force boot screen and the Kikgen logo flashes up occasionally. Output from `ssh root@xxx.xxx.xxx.xxx 'cat /media/TKGL_BOOTSTRAP/tkgl_bootstrap_[ProjectData]/doer_list'` is 
`TKGL_DOER="playlogo"
TKGL_DOER="playlogo iamforce2@LPMK3"`

smnkrt, 2023-06-17:
👋 First thing - awesome work on MPC hacks! 🙇 

Other than that i've tried testing iamforce and I've also encountered this issue on Live 2 Retro.
It was running the 2.11.7 SSH firmware before, i've updated it via the image on the SD card and then removed the image from SD card as suggested.

`doer_list` content
```bash
TKGL_DOER="playlogo"
TKGL_DOER="playlogo iamforce2@NONE"
```

`top` output:
```
Mem: 326144K used, 1741092K free, 5900K shrd, 6904K buff, 64460K cached
CPU:   1% usr  48% sys   0% nic  49% idle   0% io   0% irq   0% sirq
Load average: 3.28 1.74 0.70 5/165 2474
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1950     1 root     R    42468   2%  25% (fstrim)
    1     0 root     R    46292   2%  25% {systemd} /sbin/init
  303     1 root     S     148m   7%   0% /bin/DeviceControlServer.bin DeviceID="DeviceID01020304E8090000534D0000DECAFBAD00000000000000000000000000000001"
 1020   974 root     R     2792   0%   0% top
   15     2 root     SW       0   0%   0% [rcuc/0]
   23     2 root     SW       0   0%   0% [ksoftirqd/1]
  104     2 root     IW<      0   0%   0% [kworker/u9:1-br]
  330     1 root     S    29216   1%   0% /usr/libexec/udisks-daemon --no-debug
  181     1 root     S    19804   1%   0% /usr/lib/systemd/systemd-udevd
  269     1 systemd- S    19492   1%   0% /usr/lib/systemd/systemd-timesyncd
  169     1 root     S    14788   1%   0% /usr/lib/systemd/systemd-journald
```

Calling `reboot` hangs - if that might be a clue.

@TheKikGen if you have some debugging tips/commands i'd gladly run them to try to figure this out.