TheKikGen/MPC-LiveXplore #50: No way to scp Arp file on /usr/share/Akai/SME0/Arp Patterns/

air-v, 2023-03-31:
Hello @TheKikGen 
I'm  trying to add more arp and rack FX on MPC but it won't due to Read-only file system even i apply those command 
# systemctl stop inmusic-mpc
# mount remount -o rw,remount /

Regards

TheKikGen, 2023-04-01:
I really not recommend to modify the rootfs as it will be erase at the next update.
Use the "arp_overlay" module of the bootstrap instead.

Know that it can slow the boot process as arp in load in memory; I don't know why...
The clue is to reduce the number of arp pattern.

air-v, 2023-04-01:
Sorry, that was stupid of me... I didn't check in the right path....thanks you to remind me to check first in modules place.
Of course, you can close this message

closed 2023-06-26 by TheKikGen