TheKikGen/MPC-LiveXplore #51: Denon SC Live Signed Images

JeffWelder, 2023-04-01:
# SC Live 4 Engine DJ OS 3.0.0
I'm looked at the Denon SC Live 4, and it now looks like Denon started signing their firmware images and using sha256 rsa2048. :(

Guess they're using a newer processor with u-boot, and decided to do things the right way.

Just wanted to share in case anyone was interested.

[SC Live 3.0.0 img](https://imb-cicd-public.s3.amazonaws.com/Engine/3.0.0/Release/EOS/894d9ce1ab/SCLIVE4-3.0.0-Update.img)
  
## Had to delete prefix up to D00DFEED:
```415A3078 01000000 00000000 38000000 38000000 90000000 A8000000 E8020000 58000000 03000900 01000000 E8020000 01000000 07000000 00332E30 2E300045 6E67696E 654F5320 75706772 61646520 696D6167 65004A43 31315300 4A503230 004A5032 31007370 6C617368 00757064 61746573 706C6173 68006B65 726E656C 00726F6F 74667300 64656E6F 6E2D3100 07D0E415 1E000000 11D0E415 24000000 12D0E415 29000000 424F4F54 00000080 F0030000 00000000 ECC70800 00000000 00000000 01000000 1A7F13BE 2A4D5300 F1F0B2AD EC205D77 8E83CD7E F35D2AC8 160DC380 EAE196D4 424F4F54 00000080 E0CB0800 00000000 A4B00800 00000000 00000000 02000000 CA0804FB 9EBC15FA EAF951CE E7F36DF9 054DF6CD E4EE713A E41F61E5 90D2E95B 424F4F54 00000080 887C1100 00000000 A4B00800 00000000 00000000 04000000 27E78374 17F4D87A 9D7031C1 C2265F1C 1074958A B2C71CF5 567E704B A4FBDC5D 50415254 01000080 302D1A00 00000000 20150000 00000000 2E000000 01000000 AF9CF259 237B2DCF C5175AAF 0FD080D6 A7462637 12C7F7F6 7352F6D9 D7638913 50415254 01000080 50421A00 00000000 8C150000 00000000 2E000000 06000000 F8757CBE 2B54952B 9D157DC1 3CC881FE 9BF5E64A 3B636DC3 08033DE6 F7119F6B 50415254 01000080 E0571A00 00000000 90150000 00000000 35000000 01000000 4D337D66 3EDC2ABC E1BBE28F 69BB440F 4E5EDD32 73498D5D 7B0C5289 F7997036 50415254 01000080 706D1A00 00000000 80150000 00000000 35000000 06000000 5F9B86EA A6E8794C 67D5AF66 A6630956 3A53F7B8 5DCBE23E CBEACFBC ABE3A976 50415254 01000000 F0821A00 00000000 58C85000 00000000 42000000 00000000 E0D0793D C5154EC1 3E4E03CD 1F09BCC9 0C0B010B FB37FE54 6F61ED89 A403F782 50415254 01000000 484B6B00 00000000 5C19C409 00000000 49000000 00000000 F28C5924 279FA986 3342BB92 2C0420CA 74593A8B 400ABCDD 77BF5E13 4DB4D3C0 50000000 00000000 0170949F 14B5636B 8674368B 9B36251A CEA8A1B6 43C2ED23 98B60D5B FB529D49 275CACF2 5B5C41BB F1A679CD 7843C671 5908CB2D A0E206E6 8C70BF87 43EA3419 2133DC24 B2FB6471 45D6A7BB ECACB878 74D5B068 F1E76677 ED2A44B7 9E554EF3 2B03FF97 5D4E835E 3E388964 E34D8C37 39E857C0 A579A40A CC8AFD9A A476C28F 605BE61F 50C38C86 8E66E64D 08FEFC03 FF4D36A8 C7A45813 FBF47251 A2DA9A8D 8123DF82 78207CE4 BA255643 8B581AEC 2FC819F0 9CB46E02 438A0478 88835902 35790CCA AFBE8DAB E51D8C80 2416FAC8 2BEEEC6A 32C465A8 EBB5E40E 0BDB669C 3DC6B8F3 3093CE6A A0771860 6CD0F91E C473CFEF DD2707DF DCFC5596 23A366B6```

---

`brew install u-boot-tools`
`dumpimage -l modSCLIVE4-3.0.0-Update.img`
```FIT description: bootupd
Created:         Fri Feb 24 08:44:41 2023
 Image 0 (idbloader)
  Description:  idbloader
  Created:      Fri Feb 24 08:44:41 2023
  Type:         Rockchip SD Boot Image
  Compression:  uncompressed
  Data Size:    114688 Bytes = 112.00 KiB = 0.11 MiB
  Hash algo:    sha256
  Hash value:   a9e613b9b9b4919847292203e802d5aee615e3053976bd7aff3530ede2e5f1af
 Image 1 (u-boot)
  Description:  u-boot
  Created:      Fri Feb 24 08:44:41 2023
  Type:         Kernel Image
  Compression:  uncompressed
  Data Size:    459312 Bytes = 448.55 KiB = 0.44 MiB
  Architecture: ARM
  OS:           U-Boot
  Load Address: unavailable
  Entry Point:  unavailable
  Hash algo:    sha256
  Hash value:   35a48bdaced75921200dc5a16f5c8d163929448c823e76bd5a12f7bd599eda22
 Default Configuration: 'config'
 Configuration 0 (config)
  Description:  image
  Kernel:       unavailable
  Sign algo:    sha256,rsa2048:denon-1
  Sign value:   31386e7bf0ff3e422ccb4d76a0ba5b2e8a06f22238141796f1275cc7b3ed6100690c65bd6d046ac72dbc6496306411922ff39f5199a730d166288f2a4066fa1f369b0e8f9eb614f87486728d51b33935dd9936be9834d417a6c1145bc7983d4ef4561d367bb5eefe2577e922928702c36831bf21bfedc9ac94b95f53efcf2eff9584a186e45e0894cc537576990254a9f7aa5db994a1e409ce251d2a9173dd1a8115be5ef5754b0cc5f9fea7ba472b5c4d805d7c9c1875308b70087c3eef4606a741021666fa635abb4a78c86e49c9eec42ae056836fe0708c26407b3a1e9e4f22c737d96326c104982f39250a276c06619723090059689d987dce2594b53896
  Timestamp:    Fri Feb 24 08:44:41 2023
  ```
`dumpimage -T flat_dt -p 1 -o uboot modSCLIVE4-3.0.0-Update.img`

`mkimage -l uboot`
```FIT description: DENON DJ PRIME 4 Plus bootloader
Created:         Fri Nov 18 10:02:41 2022
 Image 0 (u-boot)
  Description:  U-Boot for DENON DJ PRIME 4 Plus
  Created:      Fri Nov 18 10:02:41 2022
  Type:         Firmware
  Compression:  uncompressed
  Data Size:    394944 Bytes = 385.69 KiB = 0.38 MiB
  Architecture: ARM
  OS:           U-Boot
  Load Address: 0x01000000
  Hash algo:    sha256
  Hash value:   5455f40142fb7b0b891917d820ecd419e579f13f49659820179d3a87e83156b8
 Image 1 (fdt-jc11s)
  Description:  rk3288-az01-jc11s
  Created:      Fri Nov 18 10:02:41 2022
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    62736 Bytes = 61.27 KiB = 0.06 MiB
  Architecture: ARM
  Hash algo:    sha256
  Hash value:   c2fed92e270d9ca5eb19cb9644f1a6dd016811d38a01cd2a222708b95eaf276e
 Default Configuration: 'jc11s'
 Configuration 0 (jc11s)
  Description:  rk3288-az01-jc11s
  Kernel:       unavailable
  Firmware:     u-boot
  FDT:          fdt-jc11s
  Loadables:    u-boot
  Sign algo:    sha256,rsa2048:stage2-denon-1
  Sign value:   44aa0646fc743724da51b2117a89a845f2b27a2f6a14216aff9bc433276dac1d9c18deb096193e3e8b0ec13587449554592e2053a363d0390e7d739b1858430bcdecbaade9af248a4d468ec55ddfbe741e472e3f3c80f41629fdbd8b0493ca609a39d234a83bb0de0c9f9752e340c00431c8da408ee7c39a871dc9a3b8672f81b319eca99a3fde2ecac00beaa58ac9a18d337b8852df2210517afebbd06bce679f4ebdc013e95fcdf89e8416e104191a00aaee8d7071dbbf61bcb03cc91bef3b2b4ca18f53d8ef50fad426d6137f3653e1c16b691bb98dd0d9782bc4813347bfa55eed3ec5cac9e0cf0c60b6950c689f841efb3beb10dbe30f8fdb340e3fc9d6
  Timestamp:    Fri Nov 18 10:02:41 2022
  ```

JeffWelder, 2023-04-01:
Seems like I was looking at JC11s instead of JP21, but they're all in the image file.

TheKikGen, 2023-06-26:
Thanks for sharing Jeff.