TheKikGen/MPC-LiveXplore #52: Question - keyboard mapping

atalun, 2023-05-07:
Just curious if any one stumbled upon a way to map keypresses to functions. By default only "TAB" button is supported (focus on next item) and not any other. It would be great if we find a way to use usb keyboard for shortcuts in standalone. I looked trough config files and didn't find anything remotely suggesting that possibility.

TheKikGen, 2023-06-26:
Good point !
For what exactly ?

atalun, 2023-06-26:
To map additional functions to buttons. I came about to this idea when mapping my MPK249 keyboard, which has option for sending regular keystrokes vis usb. Some buttons are hardwired to sending only keypresses (however, I can decide which keypresses are being sent) and wanted to use them (especially the arrow keys). The only keypress that is received is TAB (which moves to the next item on the screen) which itself is great, but it would be even better to being able to move back:) If such keyshortcuts could be used, it would be possible to attach some usb keypad for faster function access (I would love to no have to touch the screen so much)

TheKikGen, 2023-11-30:
Continue discussion here : https://github.com/TheKikGen/MPCLiveXplore-libs/discussions/17

closed 2023-11-30 by TheKikGen