TheKikGen/MPC-LiveXplore #53: Usin APC MINI MKII FAder as volume mixer

air-v, 2023-08-21:
Hi @TheKikGen 
How have you been since last time? 
Now that I'm better at assigning keys via the .H in DEV's mmp directory, I've replaced the KIKPAD with the APC mini (which handles colors better, especially for record/mute/solo) and has 1 extra button, ideal for the Shift function.
The problem is that no matter how much I tweak the knob part into a fader, it doesn't work as it should. In fact, it increments and decrement if you move the fader, but once stuck up or down it's not really effective. I try to assign the fader value directly to the volume of the track I'm working on, even if the volume jumps from the old position to the new one, it will still be more effective live.

Thanks for your help.
air-v

TheKikGen, 2023-09-22:
Knob on Mpcs are infinite and are relative not absolute.  So if you use fader, it is by construction, absolute values.
You need to map as usual the faders (midi map).

air-v, 2023-09-22:
Hi @TheKikGen,
I know that faders and encoders don't work the same way. I thought there was a method for transforming relative position into absolute position, but I find that midi mapping is not the best solution. As this method doesn't follow what's on the pad matrix, it means you only have access to 8 fixed volume knobs. I'll try to see if such a method exists on arduino and transpose it.
I'll keep you posted.

TheKikGen, 2023-11-15:
In fact, there is no way to known the current value of a know within the MPC application, because we are outside. We don't have a direct access to the app memory.   Without that/these values in differents context, you can't synchronize  with the fader value.  The knob only sends midi message like "turn left n step", "turn right n step", "touched".