TheKikGen/MPC-LiveXplore #59: Akai Force and libinput control loop

matchaxnb, 2024-02-10:
Hey,

I'd like to write a libinput plugin for the Akai Force to map the buttons on an Akai MIDIMix to various functions such as enable ratcheting, quick jump to some menu items and so on (the MIDIMix sends cc value 127 / 0 for its buttons and may also send note on / note off). I'd also like to do that for the APC Mini MK2 that I also own, for example to extend the launch section. Is that possible according to your knowledge? I read in your docs something about the Akai Force "not being there".

If I were to work on that the most important thing would be a dump of the MPC X internal midi protocol when pressing buttons such as ratchet and other direct menu access stuff.

I already work on modern mpc stuff at the moment, specifically writing a [good AKP to XPM converter](https://github.com/matchalunatic/akai-akp-python) for old samples and keygroups and also [working on a XPJ parser and editor](https://github.com/matchalunatic/akai-xpj-tools) so we can edit stuff in MPC projects that are highly impractical to touch with the UI such as midi mappings. Bridging the button gaps would be quite liberating for my music production.

TheKikGen, 2025-04-10:
For buttons mapping, check the IamForce2 project.
I capture the internal midi flow and sysex.
Read that too : https://github.com/TheKikGen/MPC-LiveXplore/wiki/MPC-internal-controller---Midi-aspects

TheKikGen, 2025-04-10:
https://github.com/TheKikGen/MPCLiveXplore-libs/wiki