TheKikGen/MPC-LiveXplore #68: Image tool doesn't work with MPC 3 beta

fudini, 2024-10-06:
It looks like 3 beta no longer can be extracted with `mpcimg` script:

```
---------------------------------------------------------------------
AKAI MPC/FORCE IMAGE TOOL - V1.2
https://github.com/TheKikGen/MPC-LiveXplore
(c) The KikGen labs.

NB : the dtc utility must be acessible from the current path !
---------------------------------------------------------------------

Akai image [mpc-3.4.0.92BETA-update.img] details :

Found properties : [] 

hash value           : 

Done !
```

Just running `fdtget` gives:
```
Error at '/': FDT_ERR_BADMAGIC
```

fudini, 2024-10-22:
@simonvpe thanks for the script, I will definitely test it later!

simonvpe, 2024-10-24:
Just a note to the maintainers of this project:

The rootfs is no longer contained in a dtb. In the 3.4beta3 image there's first a chunk of 0x152 bytes, then the rootfs xz compressed. So if the maintainers of this project want to use my script I recommend to instead of using binwalk to scan the image for the first xz header to find the start offset, then parse out the length, then use dd to extract the rootfs.xz. Binwalk is quite a hefty dependency and I simply wanted something quick and dirty to get ssh up, to try and troubleshoot my audio interface problems.

Another possibly important piece of information is that I used the image bundled inside of the MacOS installer app. While unlikely, it's possible that the images are put together differently depending on how they are bundled with their installers.

gatesphere, 2024-11-27:
> Another possibly important piece of information is that I used the image bundled inside of the MacOS installer app. While unlikely, it's possible that the images are put together differently depending on how they are bundled with their installers.

FWIW, the update.img for 3.0 Beta 4 has the same sha1sum for both the USB .img package and when extracted from the Windows updater, so I don't believe the images are being generated differently.

Is it possible to build and flash an unsigned image in the same old way as before?  Your post seems to indicate so.

simonvpe, 2024-12-11:
I flashed the uncompressed image using fastboot, and it does indeed work.

TheKikGen, 2025-03-09:
See the new mpcimg2 tool for the 3.4 image format

tertiumndatur, 2025-05-27:
> I flashed the uncompressed image using fastboot, and it does indeed work.
I tried booting the patched rootfs via fastboot on Headrush Core and it didn't work for me
https://github.com/TheKikGen/MPC-LiveXplore/issues/61#issuecomment-2002035351