TheKikGen/MPC-LiveXplore #69: mcpimg does not work with HeadRush Prime/Core firmware Update.img

quiknick, 2024-10-08:
mcpimg does not work with HeadRush Prime/Core firmware Update.img.  It works fine with mx5 firmware and I have dtc installed.
```
---------------------------------------------------------------------
AKAI MPC/FORCE IMAGE TOOL - V1.2
https://github.com/TheKikGen/MPC-LiveXplore
(c) The KikGen labs.

NB : the dtc utility must be acessible from the current path !
---------------------------------------------------------------------

Akai image [Update.img] details :

Error at '/': FDT_ERR_BADMAGIC
Error at '/images/rootfs/hash': FDT_ERR_BADMAGIC
hash value           : 

Done !
```
Works fine with mx5 firmware
```
---------------------------------------------------------------------
AKAI MPC/FORCE IMAGE TOOL - V1.2
https://github.com/TheKikGen/MPC-LiveXplore
(c) The KikGen labs.

NB : the dtc utility must be acessible from the current path !
---------------------------------------------------------------------

Akai image [Update.img] details :

timestamp            : 648069c1
description          : HG04 upgrade image
compatible           : inmusic,hg04
inmusic,devices      : 763301a
hash value           : 33ce48ac 881a4dd1 69276c8d 53d4ecf6 b11811b1

Roofs image is compressed with xz.

Done !
```

quiknick, 2024-10-12:
The FCC ID for HeadRush Prime is Y4O-HG06. HG06 is listed under InMusic

OET Exhibits List

13 Matches found for FCC ID Y4O-HG06

View Attachment	Exhibit Type	Date Submitted to FCC	Display Type	Date Available
[Declaration of authorization](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121857)	Cover Letter(s)	09/19/2022	pdf	09/20/2022
[Request for LT or STC confidentiality](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121858)	Cover Letter(s)	09/19/2022	pdf	09/20/2022
[Attestation Letter](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121859)	Cover Letter(s)	09/19/2022	pdf	09/20/2022
[External Photos](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121905)	External Photos	09/19/2022	pdf	03/19/2023
[ID Label/Location Info](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121908)	ID Label/Location Info	09/19/2022	pdf	09/20/2022
[Internal Photos](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121906)	Internal Photos	09/19/2022	pdf	03/19/2023
[MPE](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121910)	RF Exposure Info	09/19/2022	pdf	09/20/2022
[Anetnna Spce](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121902)	Test Report	09/19/2022	pdf	09/20/2022
[Test Report-1](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121903)	Test Report	09/19/2022	pdf	09/20/2022
[Test Report-2](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121904)	Test Report	09/19/2022	pdf	09/20/2022
[Test Setup Photos](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121907)	Test Setup Photos	09/19/2022	pdf	03/19/2023
[Safety Warranty Manual](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121869)	Users Manual	09/19/2022	pdf	03/19/2023
[Users Manual](https://apps.fcc.gov/eas/GetApplicationAttachment.html?id=6121909)	Users Manual	09/19/2022	pdf	03/19/2023

TheKikGen, 2025-03-06:
Try with the new utility mpcimg2.  New image format since v3.4 beta.