TheKikGen/MPC-LiveXplore #72: systemctl stop inmusic-mpc doesn't work with firmware 3.4.2

F-Thom, 2025-03-08:
I'm trying to copy new arpeggio patterns on my MPC Live 1 firmware 3.4.2 and I'm facing following issue:
The command ssh root@(Your MPC Ip addr) systemctl stop inmusic-mpc doesn't work and indeed I can't see this application inmusic-mpc in the list of the running apps.
Did something change with the firmware 3.4.2 in comparison with older versions like 2.15.x?

Using systemctl I can see acvs.service which I can stop and which indeed kills the MPC but I can't restart it (so device is dead and I've to reboot it. And additionally it seems that I don't have access to the folders to copy my midi files which are still "open". Moving from read-only to read-write and back seems to work but there's no confirmation on the command line.

Is there something I could do to copy the files I want before rebooting?
My Linux knowledge is limited, I do of course understand the commands but need simply support in this kind of cases.

I thank you in advance for your help

F-Thom, 2025-03-08:
A few additions after further investigations:

- in firmware 3.4.2, the MPC application is indeed avcs.services which can be stopped and restarted when done correctly
- the rw / ro remount worked
- I have found my solution to copy: it's the scp option -O which has to be used.
I found this advice here and it worked for me: https://stackoverflow.com/questions/26346277/scp-files-from-local-to-remote-machine-error-no-such-file-or-directory

=> So far I could manage my copy, only the name of the application to be stopped and restarted shall have to be updated (added: inmusic-mpc for firmware < 3.x, avcs.service for firmware > 3.x)

Thanks again for your great work and the information available here

TheKikGen, 2025-03-22:
Since V3, use "systemctl start acvs.service"